EmuELEC/EmuELEC #379: emuelec 4.0 test

panchaboy, 2020-12-26:
i have emuelec 3.9 (32 bits) almost every n64 roms works well with some tweaking.... naomi and atomisnwave games   like metal slug 6.. king of fighters xvi...Samurai shadown 6  run slow with coppy sound....i tried emuelec 4.0 test version (64 bits)... and those naomi atomiswave runs well with no frame drops..... but now n64 almost every game run slow 😪
tried both 4.0 version with no luck in n64 games........tried n64 directly on android system and it works great... but naomi with flycast dorectly o android works slow... lol so whats the best option?? .

i have s905x3

Maldito-1, 2020-12-31:
hey panchaboy, where did you get 4.0 test version? I would like to try with some of my tv box.

dm2912, 2020-12-31:
> but now n64 almost every game run slow

using which core?

I just tested Mario Kart 64 on Mupen64Plus Next and was getting 60fps with no graphics issues or anything.

shantigilbert, 2021-01-02:
Try latest test release: https://github.com/EmuELEC/EmuELEC-tests/releases/tag/4.0-TEST-01012021

shantigilbert, 2021-02-05:
No reply so closing this, feel free to reopen if you have more information

closed 2021-02-05 by shantigilbert