EmuELEC/EmuELEC #382: I optimized the carbon theme for the Odroid Go Advance

mtoensing, 2020-12-29:
I changed the font size to a more reasonable size and removed the description text and added the genre. Some logos got reworked for the small screen and I added the European snes logo. Maybe this is useful for someone? 

[image]

https://github.com/mtoensing/es-theme-EmuELEC-carbon-odroid-go-advance

shantigilbert, 2020-12-29:
Thanks I am sure someone will find it useful. 

Do you want me to add it to the theme downloader ?

mtoensing, 2020-12-29:
If you tell me how I do that I would gladly add it.

mtoensing, 2020-12-29:
Ah, o

closed 2020-12-29 by mtoensing

shantigilbert, 2020-12-29:
> 
> 
> Sorry, I misread the answer. Of course. Please do. How can I update it after that? Do I just push new commits into my repository?

Yes, the script pulls "master" branch. _

mtoensing, 2020-12-29:
Perfect. Thank you. If I would like to change the displayed name of my theme? Is this the name of the repository? Can I change that and your scripts adapt that? 

[image]

shantigilbert, 2020-12-29:
In that case I have to change it, I usually put the name of the repo but in theory it could be anything

mtoensing, 2020-12-29:
Ok, I will rename the repo and post the new url here. That was just a working title, but I would like to rename it if that is ok.