EmuELEC/EmuELEC #383: nintendo ds Roms

panchaboy, 2020-12-30:
i put Nintendo ds Roms in nds folder... and it wont show any emulator at start.. like its recognizing roms
im using version 4.0... (3.9 it was detecting and showing emulator)... its just with nds... all otherbsystems are fine.... any help?

SamuEDL, 2020-12-30:
Good Morning. The NDS emulator is not pre-installed in EmuELEC so I think it may give you this error. To install it, you must be connected to the Internet and access the "Setups script" section. And press "install drastic", then the NDS emulator will be installed. After installing it, check if you can already run the games.

closed 2020-12-30 by shantigilbert

LuisFPR, 2024-05-15:
> Good Morning. The NDS emulator is not pre-installed in EmuELEC so I think it may give you this error. To install it, you must be connected to the Internet and access the "Setups script" section. And [...]

Easy fix, thank you for this!