EmuELEC/EmuELEC #386: OGA and OGS Wifi Dongle Support for 5Ghz?

mtoensing, 2021-01-01:
Hi, 

I would like to prepare for the OGS and I own a OGAB. I would like to use my 5Ghz network. Is there a product everyone can recommend because it just works? I ordered some wifi dongles to test this out but do I have to deactivate the build in networlk device in my OGBA? If so how? Maybe a chapter in the wiki would be great for everyone since this question will come very often after the arrival of the OGS

shantigilbert, 2021-01-07:
I do not have recommendations, but basically any dongle that works in linux should in theory work

mtoensing, 2021-01-08:
I tested this Wifi Dongles and they all work plug'n'play with EmuELEC on the Odroid Go:

- [Plugable USB 2.0 Wireless N 802.11n](https://marc.tv/out/ogwd) (2,5 Ghz and 4,8 MB/s like build-in Odroid Wifi
- [TP-Link TL-WN725N Nano](https://marc.tv/out/wifi150) (2,5 Ghz and 4,8 MB/s like build-in Odroid Wifi
- [Asus USB-AC53 Nano AC1200](https://marc.tv/out/wifi300) (5Ghz but limited to 10 MB/s because of the USB port speed of the Ordroid GO Advance)

shantigilbert, 2021-02-08:
Closing this as resolved, thanks

closed 2021-02-08 by shantigilbert