EmuELEC/EmuELEC #393: Fine-tuned scraper configuration?

mtoensing, 2021-01-05:
Hi, 

I enjoy EmuELEC on my Odroid and I played through many games in hardcore mode the past few days. I discovered that the build in Scraper is actually pretty descent and works fine. Normally I use an external program to do that. I miss some configuration in the build in scraper like: 

Priority of the region. e.g use always US and not JAP and so on. Is there a (hidden) configuration file for the scraper where I can set parameters?

shantigilbert, 2021-01-19:
Do you mean you scrape a game and it always shows the JAP version?

mtoensing, 2021-01-19:
Yes. In my case always force us or eu and never jap. It is a flag you can set.

shantigilbert, 2021-01-19:
from the source code: 
```
// Detect ROM region
		auto info = LangInfo::parse(mFileName, nullptr);
		if (!info.region.empty())
			region = info.region;

		std::string language = SystemConf::getInstance()->get("system.language");
		if (language.empty())
			language = "en";
```
This is dependent on 2 things: 

1.- The rom name
2.- What system language you have your system on.

At the moment, there is no way at the moment to override this, does not seem like it needs override, since it is all detected on what rom name you have.

mtoensing, 2021-01-19:
The scraper can be forced to use a region. Which would be an additional option or a configuration file. But worry. Imam just annoyed by the JAP covers. I can use skyscraper manually where I can set the option.

shantigilbert, 2021-01-19:
AFAIU, JAP should not be default, it should default to your system language, unless of course you have  JAP as your language, or your game has the (j) flag in the name.

shantigilbert, 2021-01-19:
Also I think this can be forced already, if you open the game options (select the game and press select on the gamepad) under Edit this game's metadada you should see a REGION option

mtoensing, 2021-01-19:
I will try that. Thank you and keep up the good work! :)

closed 2021-01-19 by mtoensing