EmuELEC/EmuELEC #394: Duckstation crash

panchaboy, 2021-01-06:
im on emuelec 4.0 it has duckstation core for PSX games... bur every game crash.... any suggestions?
I'm supposed to have the bios but I don't know if they're the right ones.... I want to use duckstation because it has better graph performance

please help! ✌️

SamuEDL, 2021-01-07:
Good Morning, Can you tell which version of emuelec you use? (Ng, oga or normal). And a log would also help us to find the error, if you do not know how to obtain it. 
Here is a guide: https://github.com/EmuELEC/EmuELEC/wiki

shantigilbert, 2021-01-07:
Please use the issue template, pretty hard to help with no information

0kti, 2021-01-10:
I have the same - OGA

shantigilbert, 2021-01-16:
Closing this, since no response, feel free to reopen with the required information.

closed 2021-01-16 by shantigilbert