EmuELEC/EmuELEC #395: Flycast crash

panchaboy, 2021-01-07:
cant Modify TEXTURE UPSCALING (xBRZ) to any value because it crashes... it has to be off.... (tryed in all 4.0 versions).... no problem with 3.9

SamuEDL, 2021-01-07:
Good Morning, Could you send us a log to check it? Here is a guide to obtain it: https://github.com/EmuELEC/EmuELEC/wiki

shantigilbert, 2021-01-07:
Please use the issue template, pretty hard to help with no information

emutronic, 2021-01-13:
yes i have same bug ...heres my log when trying to upscale texture 

PLATFORM: atomiswave
ROM NAME: /storage/roms/atomiswave/samsptk.zip
BASE ROM NAME: samsptk.zip
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/atomiswave/samsptk.zip
2nd Argument: -Patomiswave
3rd Argument: --core=flycast
4th Argument: --emulator=libretro
Full arguments: /storage/roms/atomiswave/samsptk.zip -Patomiswave --core=flycast --emulator=libretro --controllers=-p1index 0 -p1guid 030000006d0400001dc2000014400000 
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/flycast_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/atomiswave/samsptk.zip
Emulator Output is:
[INFO] RetroArch 1.9.0 (Git 7b90da4af7)
[INFO] [Overrides]: Redirecting save file to "/storage/roms/atomiswave/samsptk.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/atomiswave/samsptk.state".
[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Jan  1 2021
[INFO] Version: 1.9.0
[INFO] Git: 7b90da4af7
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [CORE]: Loading dynamic libretro core from: "/tmp/cores/flycast_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at /storage/.config/retroarch/config/Flycast/Flycast.cfg.
[INFO] [Overrides]: No content-dir-specific overrides found at /storage/.config/retroarch/config/Flycast/atomiswave.cfg.
[INFO] [Overrides]: No game-specific overrides found at /storage/.config/retroarch/config/Flycast/samsptk.cfg.
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: remap directory: /storage/.config/retroarch/config/remappings
[INFO] [Remaps]: core-specific remap found at /storage/.config/retroarch/config/remappings/Flycast/Flycast.rmp.
[INFO] [Overrides]: Redirecting save file to "/storage/roms/atomiswave/samsptk.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/atomiswave/samsptk.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: SET_KEYBOARD_CALLBACK.
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [CONTENT LOAD]: Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/storage/roms/bios".
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: SET_SAVE_STATE_IN_BACKGROUND: yes.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[... 288 lines cut ...]
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
/usr/bin/retroarch: symbol lookup error: /tmp/cores/flycast_libretro.so: undefined symbol: omp_get_num_procs
exit 127

fabioL78, 2021-01-13:
Same here, same /usr/bin/retroarch: symbol lookup error: /tmp/cores/flycast_libretro.so: undefined symbol: omp_get_num_procs
exit 127 error. 
I'm on Odroid N2+ with the 4.0 01012021 test. Also can confirm it worked up to 3.9.

emutronic, 2021-01-15:
@shantigilbert can you please be a little more specific where to disable that feature??... i hve s905x3 with latest 4.0 test... thanka a lot

shantigilbert, 2021-01-16:
this is a compilation issue, you cannot change that in settings, need to wait until a new version is released.

emutronic, 2021-01-16:
> this is a compilation issue, you cannot change that in settings, need to wait until a new version is released.

its already fixed.??? . need to RE-Download+

emutronic, 2021-01-17:
> this is a compilation issue, you cannot change that in settings, need to wait until a new version is released.

what do you mean with FIXED_IN_DEV??...Thanks

lethal-guitar, 2021-01-17:
@emutronic it means that changes were made to the source code on the `dev` branch to fix the issue. The next version that gets built and released will include that fix. But until that happens, you'll need to wait (unless you want to compile an image yourself from the dev branch, if you have coding experience).

You could also switch to the latest stable release (3.9) which I believe doesn't have this problem.

commit 537d26f0 (2021-01-14): Flycast: Disable HAVE_OPENMP

This fixes #395
closed 2021-02-09 by shantigilbert