EmuELEC/EmuELEC #397: Mupen64plus core (not Mupen64Plus-next) doesnt work in latest 4.0 TEST version

mrabssalom, 2021-01-07:
**Describe the bug**
Mupen64Plus-Next works fine, but in 3.9 im using Mupen64Plus for my Amlogic s905x3 because i have much better perfomance. In 4.0 when you change the emulator from auto to Mupen64Plus, retroarch gives an error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any n64 rom and change emulator from auto to Mupen64Plus
2. Game wont launch and an error screen is showed.

**Expected behavior**
Rom is launched under Mupen64Plus core.

**EmuELEC version:**
v4.0-TEST-01012021

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: S905x3
 - Brand: Vontar
 - Model x3

Kenna1991, 2021-01-07:
Hi,

Im also have some poor performance on N64 with my X96 Max Plus S905X3. Any tips?

shantigilbert, 2021-01-07:
Enable thread rendering (For Mupen64Plus-next)

panchaboy, 2021-01-07:
i got better performance with emuelec 3.9 on n64 emulation..... mupenplus64- next very poor performance.

mrabssalom, 2021-01-08:
Same as me. I tweaked settings for Mupen64Plus Next and i have worse emulation by far.

mrabssalom, 2021-01-08:
> Enable thread rendering (For Mupen64Plus-next)

Already did but emulation is still laggy and games are unplayable.

shantigilbert, 2021-01-08:
Honestly this reports does not help to figure out the issue, it is so vague. Unless you guys provide more info, this will remain unanswered.

shantigilbert, 2021-01-08:
I switched to the 32bit version of Mupen64plus, I also reverted back Mupen64plus-next to an earlier version, as it seems newer versions are very slow. Will be available on next test. Meanwhile if you want to test theNEXT core unzip this file into the Cores share on samba (or /tmp/cores using SCP)
[mupen64plus_next_libretro.zip](https://github.com/EmuELEC/EmuELEC/files/5789656/mupen64plus_next_libretro.zip)

mrabssalom, 2021-01-09:
> I switched to the 32bit version of Mupen64plus, I also reverted back Mupen64plus-next to an earlier version, as it seems newer versions are very slow. Will be available on next test. Meanwhile if [...]

You are the best!

emutronic, 2021-01-17:
> I switched to the 32bit version of Mupen64plus, I also reverted back Mupen64plus-next to an earlier version, as it seems newer versions are very slow. Will be available on next test. Meanwhile if [...]

can you please share NEW Flycast, Parallel 64 and Mupen64plus (not Next) please 🙏... and thanks a lot

closed 2021-02-09 by shantigilbert