EmuELEC/EmuELEC #399: Scraped Videos not Showing

quietbiker, 2021-01-08:
v4.0-TEST-01012021

Full media scraped from "screenscraper" (I have my login)
Video folder was created for the systems and .mp4 video file exist (and can be played on PC).
Checked on every available theme.
I can only see box art.
Changing Theme Configuration, Gamelist View Style to video/detailed/grid/basic/automatic does not show the downloaded video.
I can hear audio for the video..  The video is simply not displaying in any theme.

This was working in 3.9

 - SOC: [ S095x3]
 - Brand: [e.g. Beelink]
 - Model T95 Max+
 - 256GB Card

shantigilbert, 2021-01-08:
Download the test again

`EDIT: Re-uploaded due to a small bug, even though the build date has changed it should still be considered v4.0-TEST-01012021`

closed 2021-01-08 by shantigilbert

quietbiker, 2021-01-08:
Thanks for the reply.  I did download and flash this yesterday evening - sorry to ask, but has it changed in the last 36hrs?

shantigilbert, 2021-01-08:
Not, but that bug is exactly what was fixed, unless I somehow missed the generic version