EmuELEC/EmuELEC #401: NFS Support

NekoEd, 2021-01-10:
**Is your feature request related to a problem? Please describe.**
EmuELEC currently only has CIFS support for network filesystems; this is fine for networks with Windows machines/servers, but puts those of us who use NFS (and do not want to use Samba) out in the cold.

**Describe the solution you'd like**
I would request the addition of NFS support for Linux servers not running Samba due to a lack of necessity.

**Describe alternatives you've considered**
This is an alternative to the current support already present.

**Additional context**
Our network does not have Windows machines that require access to network file shares hosted by a Linux server, and I imagine this may be the case for other users as well, particularly in cases where a Samba file share is not desired and NFS is already in place. Linux has mainline support and tools for NFS and addition of support for NFS (all versions, up to 4.2) should be trivial; enabling support in the kernel and adding the NFS tools to the image should be sufficient, then instructions for mounting the filesystems is nearly identical to those for CIFS.

shantigilbert, 2021-01-16:
I have no experience, nor a NFS network anywhere near me, how could I test this?

shantigilbert, 2021-01-16:
Have you tried this? 

https://github.com/EmuELEC/EmuELEC/blob/master/packages/sysutils/systemd/config/system.d/nfs.mount.sample

NekoEd, 2021-01-22:
Ah, my apologies for the late reply. I will give that a try as soon as I
can.

shantigilbert, 2021-02-05:
Closing this as no reply for 14 days, feel free to reopen if the issue persists.

closed 2021-02-05 by shantigilbert