EmuELEC/EmuELEC #409: MAME games wont load

emutronic, 2021-01-11:
hi....im using latest 4.0 test version ...with previus 3.9 stable version all games load..i have all same bios of 3.9...maybe i need to relocate or rename something?....heres the log ....thanks for your help

PLATFORM: mame
ROM NAME: /storage/roms/mame/mame2010/1942.zip
BASE ROM NAME: 1942.zip
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/mame/mame2010/1942.zip
2nd Argument: -Pmame
3rd Argument: --core=AdvanceMame
4th Argument: --emulator=AdvanceMame
Full arguments: /storage/roms/mame/mame2010/1942.zip -Pmame --core=AdvanceMame --emulator=AdvanceMame --controllers=-p1index 0 -p1guid 030000006d0400001dc2000014400000 
Run Command is:
/usr/bin/bash /usr/bin/advmame.sh /storage/roms/mame/mame2010/1942.zip
Emulator Output is:
1-n3a.bin    NOT FOUND
1-n4.bin     NOT FOUND
1-n5.bin     NOT FOUND
1-n6.bin     NOT FOUND
1-n7.bin     NOT FOUND
1-c11.bin    NOT FOUND
1-f2.bin     NOT FOUND
2-a1.bin     NOT FOUND
2-a2.bin     NOT FOUND
2-a3.bin     NOT FOUND
2-a4.bin     NOT FOUND
2-a5.bin     NOT FOUND
2-a6.bin     NOT FOUND
2-l1.bin     NOT FOUND
2-l2.bin     NOT FOUND
2-n1.bin     NOT FOUND
2-n2.bin     NOT FOUND
08e_sb-5.bin NOT FOUND
09e_sb-6.bin NOT FOUND
10e_sb-7.bin NOT FOUND
f01_sb-0.bin NOT FOUND
06d_sb-4.bin NOT FOUND
03k_sb-8.bin NOT FOUND
01d_sb-2.bin NOT FOUND
02d_sb-3.bin NOT FOUND
k06_sb-1.bin NOT FOUND
01m_sb-9.bin NOT FOUND
ERROR: required files are missing, the game cannot be run.
Using device path: /js
Using button mode: hold
Load cfg file ...
Amlogic ini not found /storage/.kodi/addons/script.emuelec.Amlogic.launcher/config/jslisten.cfg
Amlogic-ng ini not found /storage/.kodi/addons/script.emuelec.Amlogic-ng.launcher/config/jslisten.cfg
Will Run: /usr/bin/killall advmame
button1: 8
button2: 7
exit 1

shantigilbert, 2021-01-11:
MAME ROMS with advmame will only launch from folders with no subfolders, like this:

`/storage/roms/mame/1942.zip`

I am not sure how you had your 3.9 ROMS set up but this has always been the case AFAIK

emutronic, 2021-01-11:
> MAME ROMS with advmame will only launch from folders with no subfolders, like this:
> 
> `/storage/roms/mame/1942.zip`
> 
> I am not sure how you had your 3.9 ROMS set up but this has always been the case AFAIK

its weird im using exactly the same setup with 3.9....how can i show you hows my setup??

emutronic, 2021-01-11:
[image]
[image]

this my 3.9 setup working good... can you please help with 4.0 setup please

dynath, 2021-01-15:
Move your games from the "mame2010" folder to the "mame" folder.  The Mame emulator can't read the sub-directory so it thinks the files are missing.

shantigilbert, 2021-01-17:
Like I said, advancemame, will ONLY load games that are directly in /storage/roms/mame (or /storage/roms/arcade) NO subfolders, I have no idea how it worked for you on 3.9, but I have a suspicion that you were not using advancemame before. 

Closing this, as this is not an issue, but user error.

closed 2021-01-17 by shantigilbert