EmuELEC/EmuELEC #410: Micro-sd : Error in mount_flash: mount_common: could not mount LABEL=EMUELEC

HugKR, 2021-01-13:
I'm a newbie with Emuelec, i have a Android Box TV HK1 X3 (SX905X3, 4G RAM)
I prepare correctly my micro-sd card with the release 3.9 of Emuelec with the correct img

When i put my micro-sd card in the micro-sd port of the box and boot, i've this message :

Starting debugging shell for boot step: mount_flash... Type exit to quit
Error in mount_flash: mount_common: could not mount LABEL=EMUELEC

If i put my micro-sd in a Micro SD TF Card Memory Flash Drive Adapter, and plug it
in a USB port of the box-tv, it works !?

shantigilbert, 2021-01-13:
its either a bad flash or HW related, flash your SD card again

closed 2021-01-13 by shantigilbert

HugKR, 2021-01-13:
Il'd already flash my SD card 6 times with different utility like BalenaEtcher, Rufus and Win32DiskImager, all the times get the same result.
If it was a bad flash, why it works when i put my sd card in a flash drive adapter ?

shantigilbert, 2021-01-13:
Like I said it's either a bad flash or a hardware error, you say if you use USB it works then it is you card slot, or try another card, this is not an emuelec error nor bug.

2ndMessiah, 2021-08-14:
encountered this problem today...try that tf card on Android and it couldn't be recognized either. reckoning that x96max+ doesn't support tf card >32gb??

alanbarretos, 2024-11-26:
estou com mesmo problema, mas acredito q seja o cartao , pois coloquei outro cartao na mesma entrada da tv box e funcionou,descartando problema de hardware da tv box , ele funcionou uma vez depois de restaurar a tvbox de fabrica e da primeiro boot, acessou depois de um tempinho ficou reiniciando  na tela de carregamento dos consoles e jogos.