EmuELEC/EmuELEC #413: x88 pro x3 tv box does not boot for emulec

kaderr09, 2021-01-15:
hello
i had an amazing experience with emulec with tanix tx3 mini
then a cousin of mine have asked me for a cheap tv box box for gaming , so i suggested t95 max+ but it was out of stock he ended up with x88 pro x3 tv box . an Amlogic franklin 4g/32g
i flashed a 32 gb  Sandisc micro sd with etcher, move the appropriate sm1_s905x3_4g.dtb file to the root of the sd, rename it to dtb.img .however, it failed to boot into emulec so i tried many dtb files like sm1_s905x3_4g_1gbit.dtb ...g12a_s905x2_4g.dtb and more ,all with failure and boot into android box menu 
any suggestions please 
thanks

shantigilbert, 2021-01-15:
https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install did you follow this step by step?

kaderr09, 2021-01-16:
> https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install did you follow this step by step?

thank you for the suggestion , i had an experience like i have said with tx3 mini and it was great thanks to your support man . so , i have followed the steps excepting terminal emulator . will have a try thanks

kaderr09, 2021-01-16:
i have used terminal emulator and the result is sm1_ac213_2g !!
haven't found a dtb like this , and when i try  sm1_s905x3_4g and reboot using terminal emulator is stuck on the entering logo of the tv box

shantigilbert, 2021-01-16:
If there is no dtb for that machine I am afraid there is not much you can do at the moment.

kaderr09, 2021-01-16:
> If there is no dtb for that machine I am afraid there is not much you can do at the moment.

so, nothing to do with the black screen !!!
anyway thank you for the support

shantigilbert, 2021-01-16:
Well if you have tried all the sm1 DTBs, yeah there is nothing to do at the moment, the DTB is not there. I do not have the device (and not going to buy it) to properly test this.

shantigilbert, 2021-01-16:
maybe try the sm1_s905x3_2g.dtb

closed 2021-01-16 by shantigilbert

kaderr09, 2021-01-16:
> maybe try the sm1_s905x3_2g.dtb

exactly what i have just tried , it boots to emuelec and create partitions then freezes at emuelec looding page for more than 20 mins

tuyennn, 2021-06-12:
Have same issue, I play the game normally but today It was stuck on loading. EmuELEC was installed on internal

guyy140, 2024-01-08:
Hi I know it's been a long time
  But I also have a TV BOX x88 pro x3 like this and I wanted to install the emulec software on it
Is there a way to do this today and have it work for me?