EmuELEC/EmuELEC #416: Extend Volume on Memory Card for Emuelec

buaydianway, 2021-01-18:
i bought a 512GB memory card to replace my 128GB.
created an image of 128GB and flash it on the 512GB
now i need to extend the volume of the 128GB partition but cannot find a solution online
windows do not recognize the partition & the extend volume option are grey out.
can any one help?

shantigilbert, 2021-01-18:
Sorry, no idea. This has nothing to do with EmuELEC other than you are using the card for it.

closed 2021-01-18 by shantigilbert

buaydianway, 2021-01-18:
> 
> 
> Sorry, no idea. This has nothing to do with EmuELEC other than you are using the card for it.

it will be good if we can have a solution to share with other emuelec users on wiki
Many user dont want to spend time on reload and rescrape all over again

shantigilbert, 2021-01-18:
Feel free to add it if you find one.