EmuELEC/EmuELEC #418: Force reboot during "emulation station" shutdown, intermittent file corruption issue

nekonyang, 2021-01-19:
Hello, my device is ODROID N2+ and emuelec v3.9 user.

I'm using a translator. Please excuse me.

There are two errors while using Emulationstation:

If you play the game and proceed with shutdown, sometimes the waiting time is longer. (more than 10 seconds)
If this problem occurs, my device will reboot unconditionally.

This issue sometimes results in critical damage to the gamelist.xml file in the roms folder. (Unrecoverable) That's why I often do backups when I edit my game list.

I really want to know how to solve this problem.

shantigilbert, 2021-01-19:
I do not know how to solve that issue, I have never had it happened to me, but if you mean you turn off the device while playing a game, I assume its normal that corruption might happen.

shantigilbert, 2021-01-19:
Closing this issue as it did not follow the template issue, please reopen when te required information

closed 2021-01-19 by shantigilbert

nekonyang, 2021-01-20:
> 나는 그 문제를 해결하는 방법을 모르고, 나에게 일어난 적이 없었지만 게임을하는 동안 장치를 끄는 것을 의미한다면 손상이 발생할 수 있다고 생각합니다.
Shut down by procedure.. Did not turn it off during the game.