EmuELEC/EmuELEC #420: NDS system is not showing in Emuelec 3.9

panvarg11, 2021-01-19:
**Describe the bug**
NDS (nintendo ds) emulator does not show in the list of game emulators on the emuelec main interface.

**To Reproduce**
Steps to reproduce the behavior:
1. Add NDS roms to ROMS>NDS (using execute command or directly  into memory folder if using linux OS)
2. Start emuelec in the device 
3. Look for Nintendo DS
4. Nintendo DS game list does not show.

**Expected behavior**
Nintendo DS game list should show up and allow you to select a game to play as worked in a previous version ( eg: emuelec 3.5)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version  3.9]

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: S905W
 - Brand: MyMobile
 - Model X96mini
 - Version [e.g. 22]

**Additional context**
https://youtu.be/6hiM15LV7J8
The above video shows how the issue occurs.

shantigilbert, 2021-01-19:
This is not a bug, you either used an unofficial version of EmuELEC 3.5 or you installed Drastic and don't remember, but Drastic has never been included by default, you need to install it from the setup menu entry.

https://github.com/EmuELEC/EmuELEC/wiki/Supported-Platforms-And--Correct-Rom-Path

panvarg11, 2021-01-19:
Very much Appreciated,
Correct, the previous emuelec version I found it from another website. 
Now I installed Drastic and it runs fine. 
Thank you so much!

closed 2021-01-19 by panvarg11