EmuELEC/EmuELEC #421: Shutdown System does not work, EmuElec reboots instead of shutdown

ivan-yu, 2021-01-20:
**Describe the bug**
After playing any game on Arcade emulator, go back to main menu, shutdown system.  System reboots instead of shutdown.

**To Reproduce**
Steps to reproduce the behavior:
1. Play any game on Arcade emulator, MaMe2010 or any other emulator is fine.
2. Go back to main menu, select Main Menu / Quit / Shutdown System, system appears to have gone into shutdown, but in fact, it performs reboot instead of shutdown.

**Expected behavior**
System should shut down, not reboot.

**EmuELEC version:**
 - Version 3.9

**Device**
 - SOC: S922X CPU
 - Brand: Odroid
 - Model: N2+
 - Version 4G Ram

**Additional context**
I have CorecElec 9.2.5 on eMMC, EmuElec 3.9 on microSD, PetitBoot 20201020 on SPI.  System is Odroid N2+.
I have two 2.4GHz USB wireless dongles for my two controllers for EmuElec.

shantigilbert, 2021-01-20:
Is it a clean install or upgrade?

ivan-yu, 2021-01-20:
It is a clean install, EmuELEC-Amlogic-ng.arm-3.9-Odroid_N2.img

fabioL78, 2021-01-21:
I have the same setup with the latest 4.0 test, also have the same problem, only intermittently.

ivan-yu, 2021-01-21:
If I simply start EmuElec and do not start any emulator at all, then choose shutdown immediately, it can shut down properly.

After running emulator then quit back out to main menu, and choose shutdown, then it will restart instead of shutdown.

I am suspecting whether it has something to do with USB dongle waking up the system, just a thought.

dm2912, 2021-01-21:
> I am suspecting whether it has something to do with USB dongle waking up the system, just a thought.

or CEC

shantigilbert, 2021-01-21:
Can anyone ssh to the device and type this:

```
systemctl stop emustation
small-cores enable
systemctl reboot
```
see if it reboots correctly? 

Then

```
systemctl stop emustation
small-cores enable
systemctl shutdown
```
see if it shutdowns correctly?

ivan-yu, 2021-01-22:
It seems there is no file named small-cores

EmuELEC:~ # small-cores enable
-sh: small-cores: command not found

EmuELEC:~ # find / -name small*
/storage/.config/emulationstation/themes/es-theme-EmuELEC-carbon/art/systemsbk/smallbox.xml
/storage/.config/emulationstation/themes/es-theme-EmuELEC-carbon/art/systemsbk/smallscreen.xml
/storage/.config/emulationstation/themes/es-theme-EmuELEC-carbon/art/systemsbk/smallvideo.xml
/sys/module/di/parameters/small_local_mtn
/sys/module/di/parameters/small_local_mtn1
/usr/config/emulationstation/themes/es-theme-EmuELEC-carbon/art/systemsbk/smallbox.xml
/usr/config/emulationstation/themes/es-theme-EmuELEC-carbon/art/systemsbk/smallscreen.xml
/usr/config/emulationstation/themes/es-theme-EmuELEC-carbon/art/systemsbk/smallvideo.xml
/usr/share/mc/syntax/smalltalk.syntax
EmuELEC:~ #

ivan-yu, 2021-01-22:
> > I am suspecting whether it has something to do with USB dongle waking up the system, just a thought.
> 
> or CEC

I did not turn off anything along the HDMI chain when I experience this issue, so in my case, it cannot be due to CEC

shantigilbert, 2021-01-22:
Sorry the command should be `emuelec-utils small-cores enable`

ivan-yu, 2021-01-22:
Reboot works

Shutdown does not.  `systemctl shutdown` gives unknown operation.
EmuELEC:~ # systemctl stop emustation
EmuELEC:~ # emuelec-utils small-cores enable
EmuELEC:~ # systemctl shutdown
Unknown operation shutdown.
EmuELEC:~ #

However, `systemctl poweroff` works

ivan-yu, 2021-02-14:
I tested this on the new v4.0 (bug fix updated) release, this bug is still there.

shantigilbert, 2021-02-14:
I have tested it on 4 N2 (2 regular, 2 N2+) and the bug is not here, so there must be something else going on.

ivan-yu, 2021-02-14:
Did you play a game first, then go to back main menu, then shutdown?

If I immediately shutdown after emuelec boots up, it works.

If I play some game on MaME, then exit back to main menu, then shutdown, it reboots instead of shutdown.

Thanks again.

shantigilbert, 2021-02-14:
Which version of MAME?

ivan-yu, 2021-02-14:
mame2010

shantigilbert, 2021-02-14:
You are right, after playing a game it will not power-off, I will check it, thanks for reporting

Ulisses-Medeiros, 2021-02-14:
this has also happened to me using the PCSx-Rearmed 32bit, play a game and sometimes it Reboot instead of performing the shutdown

shantigilbert, 2021-02-14:
@ivan-yu can you create a file

`/storage/.config/emulationstation/scripts/quit/quit.sh`

then type this in that file 

```
#!/bin/bash

case ${1} in
 "reboot")
    systemctl reboot 
 ;;
 "shutdown")
    systemctl poweroff
 ;;
esac
```

ivan-yu, 2021-02-14:
There is no folder named quit, create that as well?  Make quit.sh +x ?

shantigilbert, 2021-02-15:
yes, and yes

ivan-yu, 2021-02-15:
Seems to be working!

commit 3d8c5c31 (2021-02-17): emuelec-emulationstation: Correctly handle reboot and power-off 

Fixes #421
closed 2021-04-24 by shantigilbert