EmuELEC/EmuELEC #422: Hypseus fails to start on N2+ Amlogic ini not found

fabioL78, 2021-01-21:
[lair.log](https://github.com/EmuELEC/EmuELEC/files/5851669/lair.log)
Followed the info [here](https://github.com/EmuELEC/EmuELEC/wiki/How-to-play-Daphne-games-(laserdisc)-with-Hypseus) but Hypseus fails to start. From the log:
"Could not initialize window: mali-fbdev: Can't create EGL window surface
Segmentation fault
Using device path: /js
Using button mode: hold
Load cfg file ...
Amlogic ini not found "

One thing to note, the error prior to that related to missing sound files in the storage/daphne/sounds folder, which was empty. I had to download the files [here](https://github.com/btolab/hypseus/tree/13b93354792a33d09656bc197b2a9a9aef8ca8f4/sound).

EmuELEC version:
v4.0-TEST-01012021

Device
SOC: S922X CPU
Brand: Odroid
Model: N2+
Version 4G Ram

shantigilbert, 2021-02-05:
This might be related to how you have your roms setup, I could not reproduce. 

Sound files are definitely needed though as noted on the wiki 
`Finally in /storage/roms/daphne/sound you put all the sound effect files.`

shantigilbert, 2021-02-14:
Closing as no reply.

closed 2021-02-14 by shantigilbert