EmuELEC/EmuELEC #423: Add Wifi driver for RTL8812BU

ivan-yu, 2021-01-22:
**Is your feature request related to a problem? Please describe.**
I would like to request adding Wifi driver for RTL8812BU

**Describe the solution you'd like**
Wifi driver for RTL8812BU added to release 4.0

**Describe alternatives you've considered**
If there are steps I can follow, I am more than happy to add it to 3.9 myself.
I found driver here https://github.com/morrownr/88x2bu

**Additional context**
RTL8812BU provides much better speed than on 5Ghz band

shantigilbert, 2021-01-22:
I will try to add it, but I don't have a compatible dingle to test it, hopefully you can.

shantigilbert, 2021-01-22:
You do not mention what device you use, but It seems there is already a package for the device present, can you tell me what happens when you plug in your dongle?

ivan-yu, 2021-01-22:
Apology, RTL8812BU dongle actually works.

I am using Odroid N2+ and Cody WU1300S WiFi Adaptor (which uses RTL8812BU)
http://www.cudy.com/productinfo/1373461.html

When I look at the packages, I can only find RTL8812AU and RTL8822BU-aml but not RTL8812BU
https://github.com/EmuELEC/EmuELEC/tree/master/packages/linux-drivers
https://github.com/EmuELEC/EmuELEC/tree/master/packages/linux-drivers/amlogic

So I thought RTL8812BU is not supported

shantigilbert, 2021-01-22:
Ok, so it works ? can you close the issue?

closed 2021-01-22 by ivan-yu