EmuELEC/EmuELEC #425: No WiFi and Bluetooth on A95X F3 TV Box

jo-a11y, 2021-01-24:
Hello,

Thanks for all the work on EmuElec.

I'm not sure if I'm in the right place to ask this question and I apologize in advance.

I have an A95X F3 box with Amlogic S905X3 SoC and 4GB RAM.

EmuELEC works well. On the other hand, there is no bluetooth and WiFi.

Would it be possible to add these features to this box? I can help by providing data and doing tests.

And do you have any dongle to recommend?

Thanks in advance.

shantigilbert, 2021-02-05:
I do not have that box, so unless you provide more info on what WiFi chip is used, its impossible for me to guess

jo-a11y, 2021-02-06:
The WiFi/Bluetooth chip is a mediatek MT7668 RSN

shantigilbert, 2021-02-09:
Can you try v4.0?

jo-a11y, 2021-02-15:
Thank you the Wifi works perfectly

closed 2021-02-15 by shantigilbert

Amphil13, 2021-10-05:
On the device info page it says :
Network info
Loopback interface status : unknown
Ethernet connection status : down 
Wireless connection status : down
How do I fix this
I am using A95X F3 Air box with Amlogic S905X3 SoC and 4GB RAM
I need help

lcleocastro, 2021-12-14:
I still have this problem, but it's on Bluetooth, my dualshock 4 doesn't work on emuelec 4.3 but on android Bluetooth works normally and dualshock 4 was connected.

leandroksc, 2022-01-15:
Same problem here with BT on A95X F3 Air. TVBox can find BT devices, but when you try to connect, the device doesn't work and disconnects a few seconds later. 
BT works perfectly on builtin Android.

Is there a way to get it working?

Thank you guys!

LEKSUS23, 2024-03-05:
A95X F3 Air. Такая же проблема в emuelec v 4.6 . В v 4.7 блютуз-работает но тогда джойстик не работает в самих играх, а в меню emuelec работает! Я так понимаю проблема в сборке emuelec v4.6. В ней блютуз вообще не работает хотя в настройках emuelec v4.6 он включён. 

https://docs.github.com/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

LEKSUS23, 2024-03-05:
A95X F3 Air.  Same problem in emuelec v 4.6.  In v 4.7, bluetooth works, but then the joystick does not work in the games themselves, but in the emuelec menu it works!  As I understand it, the problem is in the emuelec v4.6 build.  Bluetooth does not work in it at all, although it is enabled in the settings of emuelec v4.6.