EmuELEC/EmuELEC #426: X96 mini (2g&16g) won't reboot or start up Emuelec. (tried every 905W device tree)

zensand, 2021-01-25:
Hello,

I couldn't believe this exists... and got me so exited... But after a few hours i am willing to give up.
- i did the tutorial
- i have tried every device tree. 
- i have tried older versions of emuelec
- i have tried other sd cards.

i will not reboot in emuelec and sticks/freezes at the x96 mini logo.

i hope someone can helped out and knows a solution.

thanks ;-)

zensand, 2021-02-01:
come on... no body? after 10 hours i am this close to by a super console x. Which is a andriod tx box with emeulec pre-installed.....

closed 2021-02-05 by SamuEDL

shantigilbert, 2021-02-05:
First: you can buy whatever you want with your money, but keep in mind devices that come with preloaded EmuELEC are NOT supported. 

Second: We do not have that box, if you have tried every dtb and it still didn't work then you have an incompatible box, hard for us to guess what is wrong if nobody has that box.

And third, keep in mind we do this for free on our free time, we are not required to answer in a timely fashion, we are not required to answer at all, specially if we do not have the required answer.