EmuELEC/EmuELEC #428: EmuELECT 3.9 on OGA: Scraper does not detect region and loads wrong boxarts, titles and screenshots

mtoensing, 2021-01-29:
Hi, 

I analyzed my problem from #393 a bit more. With the App "Skraper" from http://skraper.net/ I can scrape these roms and get the US and Germany Titles, Boxarts and ingame images in German:

Pokemon - Rubin-Edition (G) (V1.1) [!].gba
Pokemon - Ruby Version (U) (V1.0) [!].gba

I get these BoxArts and the Titles are correctly "Rubin-Edition" and "Ruby Edition":

[image]
[image]

In EmuELEC get emtpy Region metadata and the same scraper service in both apps (https://www.screenscraper.fr/) loads only the US versions for both games. I know this is a very minor problem and can be solved by using the external scraper. But maybe it is just a missing flag or something.

shantigilbert, 2021-01-29:
I will see what I can do, thanks for reporting

mtoensing, 2021-01-29:
Added more information in form of the gamelist.xml files.

ascarons22, 2022-10-10:
mtoensing ... Which option do you use for skraper,rasp,bathocera,generic?.
Thanks in advance.