EmuELEC/EmuELEC #431: About the latest beta version of 4.0?

ji125443, 2021-02-01:
The name of the game can not be in Chinese, for example, (Ghana fighter. Nes, ES front-end will appear：????? .nes）?

ji125443, 2021-02-01:
Online neighbors can't upload game files with Chinese names

ji125443, 2021-02-09:
[image]
[image]

ji125443, 2021-02-09:
[image]

ji125443, 2021-02-10:
-
[image]
[image]

mavisguo, 2021-03-12:
4.1 bata ng version fixed, but no ng version do not work.

SmartYou1995, 2021-03-14:
The garbled code of FAT32 file system with windows in Linux.The problem still exists，on 4.1.

SmartYou1995, 2021-03-14:
Add - O utf8 = 1 when using mount command

shantigilbert, 2021-03-14:
The release that fixes this is not even public yet, its v4.1 that is what fixed_in_dev  means and it needs a CLEAN (no update) install as it will needs to recreate the partitions

SmartYou1995, 2021-03-14:
> The release that fixes this is not even public yet, its v4.1 that is what fixed_in_dev means and it needs a CLEAN (no update) install as it will needs to recreate the partitions

Looking forward to the next new version update

shantigilbert, 2021-03-30:
https://github.com/EmuELEC/EmuELEC/commit/674ec9cbf7c6dd9a88e83535eacf944a95102c6b

ws02589111, 2021-03-31:
Thanks for yout update, after my test,  the ES front-end，ssh, samba is fixed， but the display in retroarch is not fixed.

EmuELEC version: latest dev branch: f5d53ae824b9e721b9d128ac74236f648d9699b9
SOC: S905M

To Reproduce
1.Default is English in retroarch, set it to cn, now you can see it display properly, but after you save the config and reboot, you can see the below:
[image]
2.And then , set it back to en, you can get the below:
[image]

ws02589111, 2021-04-02:
> Thanks for yout update, after my test, the ES front-end，ssh, samba is fixed， but the display in retroarch is not fixed.
> 
> EmuELEC version: latest dev branch: [f5d53ae](https://github.com/EmuELEC/EmuELEC/commit/f5d53ae824b9e721b9d128ac74236f648d9699b9)
> SOC: S905M
> 
> To Reproduce
> 1.Default is English in retroarch, set it to cn, now you can see it display properly, but after you save the config and reboot, you can see the below(ignore time on the screen):
> [image]
> 2.And then , set it back to en, you can get the below:
> [image]

@shantigilbert any idea for this？

ws02589111, 2021-04-06:
Set the retroarch theme to xmb and add the xmb_font = "/tmp/assets/xmb/monochrome/font.ttf"(the font i have replaced to yahei for test) to the retroarch.cfg, the problem in xmb is solved, but not work in ozone

commit d5c7c812 (2021-02-17): Use UTF8 as default for EEROMS

Fixes #431
closed 2021-04-24 by shantigilbert