EmuELEC/EmuELEC #439: X96 MAX 4gb

guy2010, 2021-02-05:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

guy2010, 2021-02-05:
Anyone use an X96 Max? Mine is stuck on the boot screen, any ideas?

SamuEDL, 2021-02-05:
If you do not give details, we can help you little. Do you have the correct dtb?

guy2010, 2021-02-05:
i copied the sm1.s903x3_4g.dtb file and renamed it to dtb.img

AlienatedDiego, 2021-02-06:
Is it possible that you are renaming the .dtb wrong?. Check if Windows shows the file extensions (dtb.img.img)

If this is not the problem check the soc, ram and ethernet speed to make sure you are using the correct .dtb

guy2010, 2021-02-07:
It seems to be fine. Any idea where the reset button is on the X96 Max? Maybe i need to boot holding in the reset button.

jefkestella, 2021-02-07:
I have the same issue with the x96max+ 4g/64g with s905x3 cpu.
It boots the first time ok and can start up the install of emuelec 3.9.
After setup wifi,... i quit the emuelec properly.
After then it doesnt reboot anymore and gives a black screen after the x96 logo.
I tried all the dtb files and nothing works :(
Does anyone has a working x96max plus with emuelec ?
Please let me know.
I think this tv box isn't compatible with emuelec ?

guy2010, 2021-02-07:
Mine is stuck on the boot screen.

jefkestella, 2021-02-07:
> Mine is stuck on the boot screen.

You need to use the reset button before you power up or use the libeElec app

guy2010, 2021-02-07:
Where is the reset button?

jefkestella, 2021-02-07:
> Where is the reset button?

You can hit it with a toothpick in the audio 3,5 jack connector

guy2010, 2021-02-07:
Did the LibeElec app work for you?

jefkestella, 2021-02-08:
> Where is the reset button?

The liberec app works also only the first time with the install of emuelec. Please let me know or it works with you

B3N313, 2021-02-08:
I had the same problem, use the 1ghz 2gbit image.

jefkestella, 2021-02-08:
> I had the same problem, use the 1ghz 2gbit image.
> It will work. I have also the 4gb version of the X96 MAX Plus.

Thx for the feedback. I will try it again with this dtb file. You needed the reset button the second time for start up or did he start up auto ?

B3N313, 2021-02-08:
Can't remember to be honest. I would try it with pressed reset button for the first boot.

This guide helped me, might be useful for you:

https://androidpctv.com/tutorial-emuelec-turns-your-android-tv-box-into-a-retro-console/

jefkestella, 2021-02-08:
> Can't remember to be honest. I would try it with pressed reset button for the first boot.
> 
> This guide helped me, might be useful for you:
> 
> https://androidpctv.com/tutorial-emuelec-turns-your-android-tv-box-into-a-retro-console/

Thanks for the link. I followed also this tutorial. Maybe it is the dtb file that you said that will help. Which version of emuelec you use with the x96 max plus ?

B3N313, 2021-02-08:
3.9

jefkestella, 2021-02-08:
> Can't remember to be honest. I would try it with pressed reset button for the first boot.
> 
> This guide helped me, might be useful for you:
> 
> https://androidpctv.com/tutorial-emuelec-turns-your-android-tv-box-into-a-retro-console/

Hello, i tried it again with dtb file sm1_s905x3_2g_1gbit.dtb and it doesn’t work :(

dm2912, 2021-02-08:
> Hello, i tried it again with dtb file sm1_s905x3_2g_1gbit.dtb and it doesn’t work :(

1gbit is for boxes with gigabit lan.. is yours gigabit or 100mb? a lot of them are mis-labeled either intentionally or by accident

boot into android and check its actually gigabit lan

The trouble with all of these generic X96, X95, H96 etc is they are all made by different manufacturers and most have completely different chips inside to each other. The ram, emmc, wifi chips, ethernet can vary drastically. 

Looking on the Coreelec forums a huge number of people have had problems with this box. For some its fine, for others on July 2020 firmware they have issues getting it to even boot from the SD. 

As a starting point I would recommend booting into Android, running something like Aida64 and seeing what you actually have. mainly, processor, ram, if the wifi is actually gigabit, what the android build date is or build number (coreelec users have had issues with build 1719 which is a clone, and August 2020 builds)

Long story short, these generic chinese boxes are all different, loads of clones, and first step is using Aida and seeing what you actually have and see if its one of the one that coreelec users have found wont work

dm2912, 2021-02-08:
> I have also the 4gb version of the X96 MAX Plus

there are probably 100 different versions of the x96 max plus. They are all made by different manufacturers, and contain a wide variety of chips

jefkestella, 2021-02-08:
> > I have also the 4gb version of the X96 MAX Plus
> 
> there are probably 100 different versions of the x96 max plus. They are all made by different manufacturers, and contain a wide variety of chips

To be honest i tried each dtb file for the s905x3 and nothing boots up emuelec the second time. I read that they can put info in  aida64 what they want and doesn’t mean anything 🙃

jefkestella, 2021-02-08:
> > > Can't remember to be honest. I would try it with pressed reset button for the first boot.
> > > This guide helped me, might be useful for you:
> > > https://androidpctv.com/tutorial-emuelec-turns-your-android-tv-box-into-a-retro-console/
> > 
> > 
> > Hello, i tried it again with dtb file sm1_s905x3_2g_1gbit.dtb and it doesn’t work :(
> 
> 1gbit is for boxes with gigabit lan.. is yours gigabit or 100mb? a lot of them are mis-labeled either intentionally or by accident
> 
> boot into android and check its actually gigabit lan
> 
> The trouble with all of these generic X96, X95, H96 etc is they are all made by different manufacturers and most have completely different chips inside to each other. The ram, emmc, wifi chips, ethernet can vary drastically.
> 
> Looking on the Coreelec forums a huge number of people have had problems with this box. For some its fine, for others on 2020 firmware they have issues getting it to even boot from the SD. Same with Android build 1719 which is a sign of cloned boxes.
> 
> As a starting point I would recommend booting into Android, running something like Aida64 and seeing what you actually have. mainly, processor, ram, if the lan is actually gigabit, what the android build date is or build number
> 
> Long story short, these generic chinese boxes are all different, loads of clones, and first step is using Aida and seeing what you actually have and see if its one of the one that coreelec users have found wont work

I don’t think this box is compatible with emuelec. I tried every dtb file for chip s905x3 and doesn’t work. Strange is that the first boot always works. Is it only the second time you need the dtb file ?

dm2912, 2021-02-08:
> To be honest i tried each dtb file for the s905x3 and nothing boots up emuelec the second time. I read that they can put info in aida64 what they want and doesn’t mean anything

To a certain degree, but there are certain firmwares for that box that dont work at all on coreelec for whatever reason

B3N313, 2021-02-08:
I actually was fiddling around with mine quite a lot before I got it to work. I did this firmware and android update. Had to download some Chinese Software to build a bootable image for firmware update. I had to write that image to an sd card, as the box doesn't boot from USB. I was also digging around on the CoreElec forum. Perhaps the update was necessary to get it to work. (Also watched some Russian YouTuber) It was really hard work to be honest and now I am kind of disappointed with the emulation performance...

B3N313, 2021-02-08:
My Android did not have the right updater so I had to download a stock firmware, flash it and after that I was able to run the Android firmware updater.

B3N313, 2021-02-08:
[image]

[image]
[image]

[image]
[image]

B3N313, 2021-02-08:
[image]
[image]

B3N313, 2021-02-08:
[image]

B3N313, 2021-02-08:
That's my box

dm2912, 2021-02-08:
> I actually was fiddling around with mine quite a lot before I got it to work. I did this firmware and android update. Had to download some Chinese Software to build a bootable image for firmware [...]

yeh, from what i read on coreelec forums it seems this is the solution to either the build 1719 android or a (fake?) july 2020 firmware, there's a firmware I think from September/October last year that can be flashed that makes them behave.

jefkestella, 2021-02-08:
> I actually was fiddling around with mine quite a lot before I got it to work. I did this firmware and android update. Had to download some Chinese Software to build a bootable image for firmware [...]

All that hard work for nothing... respect

B3N313, 2021-02-08:
I then downloaded a loaded image (from galisteo), which again booted without issues once I had replaced the dtb.img. And spend a lot of time setting it up. Then yesterday I was up until 3 am trying to figure out why performance on two of my favourite ps1 games was so poor that they were inplayable

jefkestella, 2021-02-08:
> I then downloaded a loaded image (from galisteo), which again booted without issues once I had replaced the dtb.img. And spend a lot of time setting it up. Then yesterday I was up until 3 am trying to figure out why performance on two of my favourite ps1 games was so poor that they were unplayable

Do you have the link for the dtb file of galisteo ? I think based on your pictures i have the same board inside 🙂

B3N313, 2021-02-08:
> Do you have the link for the dtb file of galisteo ? I think based on your pictures i have the same board inside 🙂

Galisteo did a 124 GB iso image of EmuELEC 3.9 loaded with roms and set up with the beautiful Artflix theme. The dtb is the same as in the EmuELEC release.

B3N313, 2021-02-08:
His image is available on his discord. You should first try to do flash the original firmware as mentioned above

jefkestella, 2021-02-08:
> His image is available on his discord. You should first try to do flash the original firmware as mentioned above

And how i can do that ? Do you have a youtube tutorial ?

dm2912, 2021-02-08:
> Do you have the link for the dtb file of galisteo ? I think based on your pictures i have the same board inside

but as already said, you really need to load something like Aida and see what your box is and what firmware its running. if its running 1719 then you will have to flash new firmware before you even begin. 

The "it looks like my box" route has failed, so you need to dig deeper, see what firmware you are running, see what the box is and then go from there

B3N313, 2021-02-08:
I will need to check. It's not that complicated, just difficult to get the right information

B3N313, 2021-02-08:
https://youtube.com/c/VasilyShevchuk-EvolutionTV-VS

This guy had a lot of information. Just switch on subtitles

shantigilbert, 2021-02-08:
I don't care how many issues you have please do not talk/share or in any other shape or form distribute illegal images with roms, this is the first warning for the ones that did that, next one will be a perma-ban.

closed 2021-02-09 by shantigilbert