EmuELEC/EmuELEC #442: EmuELEC 3.9 on OGA: Netplay error "A netplay peer is running a different core" but it's a cloned image and cores match.

mtoensing, 2021-02-07:
Hi! 

I am pleased with my OGA with EmuELEC 3.9! A friend and colleague of mine just bought the same OGA. Same EmuELEC, same cores, same roms. 

We tried NES (Nestopia, BattleCity) and SNES (Mario Kart) with the same Roms, but we always get the message "A netplay peer is running a different core". The Mario Kart ROM is from the NoIntro set. 

We double-checked the cores and roms. See attached screenshots. The port is open. What could be the cause of this problem? I search in google and since October 2020, many people report this problem. Is there an easy fix? We used the default settings. 

[image]
[image]

shantigilbert, 2021-02-10:
Can you test v4.0 ?

mtoensing, 2021-02-11:
I will. do I need to start netplay from the EmulationStation Menu or can I start it via RetroArch directly?

mtoensing, 2021-02-11:
It's the same with 4.0. I have a Netplay session open with a relay server. But I can switch off relay because my ports are open on my OGA EmuELEC 4.0. SNES, Mario Kart, MarcTVde Maybe someone can try to connect right now?

shantigilbert, 2022-03-19:
This should be fixed on v4.4 tested and working

closed 2022-03-19 by shantigilbert