EmuELEC/EmuELEC #443: How to play Colecovision?

roro582, 2021-02-08:
Downloaded the bios, cores, and placed machine and database files in bios but MSX works but Colecovision does not how to change autostart/autodetection of the emulator to play Colecovision?

shantigilbert, 2021-02-08:
Please reopen this issue following the proper template.

closed 2021-02-08 by shantigilbert

roro582, 2021-02-09:
Should I place this under Pull requests?

lethal-guitar, 2021-02-09:
@roro582 no, issues is correct. But there is some pre-inserted text when you open a new issue (that's called the issue template). Please read that and do what it says

roro582, 2021-02-09:
> @roro582 no, issues is correct. But there is some pre-inserted text when you open a new issue (that's called the issue template). Please read that and do what it says

I understand thank you.