EmuELEC/EmuELEC #444: Wifi ATH10k not working

Seba201x, 2021-02-08:
**Is your feature request related to a problem? Please describe.**
Wifi not working on X96 Max Plus
**Describe the solution you'd like**
The driver of the wifi chip
**Describe alternatives you've considered**
How to add a driver to compile it
**Additional context**
udevadm info
P: /devices/platform/ffe03000.sdio/mmc_host/sdio/sdio:0001/sdio:0001:1
L: 0
E: DEVPATH=/devices/platform/ffe03000.sdio/mmc_host/sdio/sdio:0001/sdio:0001:1
E: SDIO_CLASS=00
E: SDIO_ID=0271:050A
E: MODALIAS=sdio:c00v0271d050A
E: SUBSYSTEM=sdio

shantigilbert, 2021-02-08:
Please provide more information on your box, I have no idea how many X96 boxes are out there and they are ALL different! also, what version of EmuELEC are you using?

shantigilbert, 2021-02-08:
It seems that one is not supported under Kernel 4.9 which is what EmuELEC uses, so unless someone backports the driver, this is not possible at the moment.

closed 2021-02-08 by shantigilbert

eric4gith, 2021-03-05:
There are several varied spec of X96 Max+. I have one. The wifi chip is Cdtech 4761743. I should be using Qualcomm QCA6174A design. Linux driver ath10k should support the device. I have check the lastest CoreELEC 1.9 using lsmod command. There seems no ath10k driver included in the distribution.
Try compiling a Linux kernel from scratch may be fun. I should try it.

shantigilbert, 2021-03-05:
> There are several varied spec of X96 Max+. I have one. The wifi chip is Cdtech 4761743. It should be using Qualcomm QCA6174A design. Linux driver ath10k should support the device. I have checked the lastest CoreELEC 1.9 using lsmod command. There seems no ath10k driver included in the distribution.
> Try compiling a Linux kernel from scratch may be fun. I should try it.

Its not included because like I said, that driver does not work on Kernel 4.9, only on newer ones

eric4gith, 2021-03-05:
I am not familiar with the details of kernel version. By your words, it seems ath10k driver is not included in the kernel used by EmuELEC.
I found a webpage describing a situation that ath10k driver for 3.14.64 kernel does not support QCA6174A, and the writer found a way to amend the problem. The key is to download "compat-wireless-2017-09-25.tar.xz" from "http://mirror2.openwrt.org/sources/" and build it for Android 6.0.
Maybe this webpage can provide some helpful info. It's written in Chinese and have code examples and screen snapshots. I hope you can grab some idea by Google translation and guessing.

https://blog.csdn.net/u012851408/article/details/86605778

FYI.

shantigilbert, 2021-03-05:
Thanks, but I am not going to do this blindly, I do not have such chipset and doing all this work with no device is not going to be fun. Like I mentioned before if someone else wants to attempt this I am all open for pull requests.

eric4gith, 2021-03-05:
You are right. It's not practical to build and debug when there is no suitable device to test.
If fact, I think the problem should be solve from CoreELEC. It's the up-stream.