EmuELEC/EmuELEC #447: 4.0 on OGA : RetroAchievements can't be activated "unable to activate http status 403"

mtoensing, 2021-02-09:
I triple checked my credentials. They work on the website of https://retroachievements.org/ and on 3.9. I tried to spell my username lowercase. Nothing works. The result is that the switch in "game settings" -> "RetroAchievements" is switch off after this error.

[image]

mtoensing, 2021-02-09:
This error is a regression and special characters like "%" worked in 3.9 but not in 4.0

TheLastProject, 2021-02-09:
Same with the & symbol in passwords, it worked in 3.9 fine.

shantigilbert, 2021-02-10:
Confirmed

As a workaround: 

add the password directly to `/emuelec/emuelec.conf` but emulationstation needs to be stopped first 

ssh:
```
systemctl stop emustation
nano /emuelec/emuelec.conf
```
change the `global.retroachievements.password=` line and add your password (no escaping necessary)

this will be fixed on next release.

closed 2021-04-24 by shantigilbert