EmuELEC/EmuELEC #448: EmuELEC 4.0 on OGA: RetroAchievements hardcore mode can't be enabled

mtoensing, 2021-02-09:
I set Hardcore mode to enabled in the menus. The result in emuelec.conf : 

global.retroachievements.hardcore=1

But in retroarch it is not enabled. Even if I change the setting in retroarch.cfg manually it is no working. Only if i start a game and switch to retroarch by select + x I can enable hardcore mode. But it is disabled when I start the next game.

shantigilbert, 2021-02-10:
I tried this and I cannot reproduce, Hardcore mode gets enabled when I set it in ES, what else did you change? (since I notice you are using Rgui as default)

mtoensing, 2021-02-10:
I reseted 4.0 to it's defaults and its gone. Weird.

mtoensing, 2021-02-11:
I am sorry but the problem came back. After a few reboots harcore mode is switch off in retroarch but on in emulationstation. This is really bad because you never know when the hardcore mode is disabled. =(

mtoensing, 2021-02-11:
@shantigilbert  is there a way to force this setting to be enabled? This is unplayable for me at the moment.

mtoensing, 2021-02-11:
in /.config/retroarch/retroarch.cfg this line is always false if I ran into this bug.

cheevos_hardcore_mode_enable = "false" 

If I change it manually or via retroarch to true it always sets itself back to false after a new game restart

.config/emuelec/emuelec.conf shows this all the time: 

global.retroachievements.hardcore=1

mtoensing, 2021-02-11:
Ok, I read through /storage/.config/emuelec/scripts/setsettings.sh and it seems like hardcore is disabled because of netplay is active. My assumption would be that I get a notice that netplay disables cheevos all together but not only hardcore mode.

shantigilbert, 2021-02-11:
I asked you before if you changed something else, but you failed to mentioned you have Netplay enabled.
Cheevos does work with netplay, hardcore mode does not. This is not an EmuELEC issue, but a Retroarch limitation.

shantigilbert, 2021-02-11:
(pushed wrong button) Reopened issue. 

If you select hardcore mode from the Cheevos settings, while playing Netplay, does it really enable it ?

mtoensing, 2021-02-11:
If I select hardcore mode while I start a game via the frontend in retroarch they do work. But then, netplay is not active. I switch on netplay AND cheevos with hardcore mode on in the emuelect frontend. I did not know that they cancel each other out. Sorry.

shantigilbert, 2021-02-11:
Ok well at least you found the cause for the issue.

closed 2021-02-11 by shantigilbert