EmuELEC/EmuELEC #449: openbor is not working on S912

kjhjekwls, 2021-02-10:
Hi

I am using Tanix tx9s android box. (S912 CPU)

And I use Emuelec 3.9 in this device.

When I try to run OpenBOR, it's not working and following error is happened in OpenBorLog.txt

-----------------------------------------------------
Total Ram:  1824485376 Bytes (  1739 MB )
 Free Ram:   973844480 Bytes (   928 MB )
 Used Ram:      933376 Bytes (     0 MB )

debug:nativeWidth, nativeHeight, bpp, Hz  1920, 1080, 24, 60
Error: failed to create window: mali-fbdev: Can't create EGL window surface
OpenBoR v3.0 Build 0000 (commit hash: 0000000), Compile Date: Oct 13 2020

Game Selected: ./Paks/xxx.pak

FileCaching System Init......	Enabled
Initializing video............
'data/video.txt' not found.
Initialized video.............	320x240 (Mode: 0)
----------------------------------------------------------------------------

I tested with many openbor game in this device, but phenomenon is same.

I heard openbor is working properly on S905M, 

Is there solution for this issue?

shantigilbert, 2021-02-10:
Known issue, will never be fixed, sorry

shantigilbert, 2021-02-10:
S912 is EOL on EmuELEC at 3.9 the final version, many things do not work on S912, and they will not be fixed any time soon.

closed 2021-02-10 by shantigilbert