EmuELEC/EmuELEC #451: Overlays/bezels for seperate systems (idea)

Cee123, 2021-02-11:
**Is your feature request related to a problem? Please describe.**
You can add a bezel/overlay for a system, but the same bezel then gets shared amongst multiple systems and games. So when you change a bezel/overlay for a system, it changes it to that bezel all systems. You can save as a core override, but then it won't let you save configurations in RetroArch and systems that use the same core (i.e. genesis plus gx) end up with the same bezel.

**Describe the solution you'd like**
A way to customize and add a seperate bezel/overlay for each system.

**Describe alternatives you've considered**
Saving core overrides. Saving a new config. Saving game overrides (this works but you'd have to do it for each separate game and that would take forever)

**Additional context**
Previously I could do this with RetroPie and RetroArene/Ares since the configs for different systems are saved in seperate folders. I can take some screenshots and show what I mean.

It is just an idea, but is there any way to do this currently?

Cee123, 2021-02-11:
Just an example of what I mean.. how I had it set out on retroarena. Is it possible to do this on EmuElec?

[image]

[image]

[image]

[image]

[image]

[image]

shantigilbert, 2021-02-11:
That is already possible https://github.com/EmuELEC/EmuELEC/wiki/Bezels-&-Splash-screens read the manual installation part, it can use a bezel per system, per game or globally

closed 2021-02-11 by shantigilbert

Cee123, 2021-02-11:
> That is already possible https://github.com/EmuELEC/EmuELEC/wiki/Bezels-&-Splash-screens read the manual installation part, it can use a bezel per system, per game or globally

Ah, thanks for that. That's fantastic. Was looking around for that.

Cee123, 2021-02-11:
That's awesome, I was going to ask about splash screens as well. But it didn't occur to me to read the manual. /facepalm. Thanks again!