EmuELEC/EmuELEC #453: EmuELEC 4.0 on OGA: RetroAchievements pop-up is far to big (1/4 of the screen)

mtoensing, 2021-02-11:
On 3.9 the RetroAchievement pop-up was reasonable sized. Not it takes up almost one quarter of the screen. It is distracting to the point that you actually lose while getting an achievement. I made a mock up to illustrate the issue. 

[image]

This may be a result of the bigger screen of the OGS. But don't forget the people with the OGA. =) Is there a config setting I can change manually?

mtoensing, 2021-02-11:
Settings-> On-Screen Display -> On-Screen Notifications -> scale does not work because then everything is tiny. Maybe the icon of the achievement is too big or the resolution is too high?

shantigilbert, 2021-02-11:
You have to find the scaling that fits your needs, keep changing the number, or disable cheevos thumbnails

closed 2021-04-24 by shantigilbert