EmuELEC/EmuELEC #455: EmuELEC 4.0 on OGA: Script "Scan for ScummVM Games" does not work

miwasp, 2021-02-11:
**Describe the bug**
In Emulationstation, Setup Scripts, Script "16 - Scan ScummVM Games" does not work. Error is:
> EmuELEC:~ # /storage/.config/emuelec/scripts/modules/16\ -\ \ Scan\ ScummVM\ Games.sh
/storage/.config/emuelec/scripts/modules/16 -  Scan ScummVM Games.sh: line 12: syntax error near unexpected token `fi'
/storage/.config/emuelec/scripts/modules/16 -  Scan ScummVM Games.sh: line 12: `      fi'

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Emulationstation->Setup Scripts'
2. Select '16 - Scan ScummVM Games'
3. After a while you are back in Emulationstation but nothing happened

**Expected behavior**
The script should scan ScummVM games.

**EmuELEC version:**
 - Version 4.0

**Device**
 - Odroid Go Advance Black Edition

**Additional context**
Remove the "fi" on line 12 and the script works.

shantigilbert, 2021-02-11:
Removing the "fi" on line 12 makes the script work or just the error disappear? meaning does it create the .scummvm files?

miwasp, 2021-02-11:
Without the „fi“ it creates the .scummvm files.

shantigilbert, 2021-02-12:
Thank you, I have fixed this and will be available on next release

closed 2021-04-24 by shantigilbert