EmuELEC/EmuELEC #456: EmuELEC 4.0 on OGA:

gooosxxx, 2021-02-12:
Hi. Excellent work of the EmuELEC team.
In EmuELEC 4.0 on OGA, select and start does not work to exit PICO-8, SOLARUS and some PC PORTS so I have to turn off the device.
Someone else the same thing happens?
Thank you

mbroers, 2021-02-12:
Same for pico-8 for me

shantigilbert, 2021-02-12:
Those emulators all have a way to exit from their menus, hence why a "force kill" was not implemented. If you feel they are required, please post the ones that do not work for you (full list if possible)

For example, Pico-8 you have to go into options and then look for Quit Pico-8

mbroers, 2021-02-12:
Thanks shanti I looked around but am new to pico-8 and couldnt find it.

british-choi, 2021-02-14:
@gooosxxx 
Can you use OGA BE or OGA?
As far as I can see, program termination with "Hot key + Start" in OGA BE does not work properly.

shantigilbert, 2021-02-14:
> @gooosxxx
> Can you use OGA BE or OGA?
> As far as I can see, program termination with "Hot key + Start" in OGA BE does not work properly.
> 
> It looks similar to the issue below.
> #464

not the same issue, this is intentional, yours is a bug, which is also listed in the known issues: https://github.com/EmuELEC/EmuELEC/issues/463

gooosxxx, 2021-02-17:
thanks, I found the exit in the menu of each emulator.

closed 2021-02-17 by gooosxxx