EmuELEC/EmuELEC #457: Bug on drastic - emuelec

pedropereira1982, 2021-02-12:
**Describe the bug** Hello guys , i cannot install drastic on emuelec 4.0, it starts but i cannot press "yes" with the gamepad.
I have x96 mini 2g processor s905w

lethal-guitar, 2021-02-12:
Have you tried button A and button B? It's meant to be A to confirm but for some controllers it seems to be B. We're still figuring out why

lethal-guitar, 2021-02-12:
It it really doesn't work, as a workaround you can also plug in a keyboard and press space bar

pedropereira1982, 2021-02-12:
I pressed all butons and doesn work. I have a video. How can i Send it?

pedropereira1982, 2021-02-12:
https://www.dropbox.com/sh/fusf2ykcikd063q/AACTQ0uDGV_QlYjjXON9AsMXa?dl=0

pedropereira1982, 2021-02-12:
https://www.dropbox.com/s/lgvk4icol4q6vaj/20210212_202023.mp4?dl=0

shantigilbert, 2021-02-12:
The gamepad works fine on your video (you can see it opening the info popup), you are just pressing buttons wildly, press A (or B) move the cursor to the button and press B (or A) cursor should already be set in Yes, so just press A (or B) to select it, if you just want to close this window, press START

pedropereira1982, 2021-02-12:
Hello mate ty for your answer. Im not pressing wildly. The pop up appears when i press the button and disapears when i release. I dont have curson on screen. When i press yes nothingh happens. I cant press install on the pop up because when i release the button it disapears.

lethal-guitar, 2021-02-12:
@pedropereira1982 when you open the "Install Drastic" script, the yes button is highlighted initially. It looks like this:

[image]

You need to press A (or B) while the button has that little light blue outline. Or maybe another button, but the one that makes the pop-up appear in your video is not the right button.

pedropereira1982, 2021-02-12:
I tried every button.

shantigilbert, 2021-02-12:
Open the script, do not move anything, just press A, nothing happens? press Start to close the window, open the it again, do not press nothing, just press B.

pedropereira1982, 2021-02-12:
I press A nothing happens. I press B nothing happens. I press Y it shows pop up. I release Y pop up desapears. I press all the other buttons nothing happens. When i press R2 exits setup to games menu.

pedropereira1982, 2021-02-13:
Is there anyway to install drastic manually?

british-choi, 2021-02-13:
@pedropereira1982 
Which gamepad are you using?
I think your game controller DB is not correct.
If you can, make sure your gamepad is defined correctly in /storage/.config/SDL-GameControllerDB/gamecontrollerdb.txt

lethal-guitar, 2021-02-13:
@pedropereira1982 

> Is there anyway to install drastic manually?

See what I wrote above:

https://github.com/EmuELEC/EmuELEC/issues/457#issuecomment-778420223

pedropereira1982, 2021-02-13:
https://www.dropbox.com/s/ezcblsop4ahtlm3/20210213_111111.jpg?dl=0

# Game Controller DB for SDL in 2.0.9 format
# Source: https://github.com/gabomdq/SDL_GameControllerDB

# EmuELEC extra gamepads
190000004b4800000010000000010000,GO-Advance Gamepad,platform:Linux,a:b0,b:b1,x:b3,y:b2,back:b10,start:b15,leftstick:b12,rightstick:b13,leftshoulder:b4,rightshoulder:b5,dpup:b6,dpdown:b7,dpleft:b8,dpright:b9,leftx:a0,lefty:a1,lefttrigger:b11,righttrigger:b14,

190000004b4800000010000001010000,GO-Advance Gamepad (rev 1.1),platform:Linux,x:b2,a:b1,b:b0,y:b3,back:b12,guide:b13,start:b17,dpleft:b10,dpdown:b9,dpright:b11,dpup:b8,leftshoulder:b4,lefttrigger:b6,rightshoulder:b5,righttrigger:b7,rightstick:b16,leftx:a0,lefty:a1,

190000004b4800000011000000010000,GO-Super Gamepad,platform:Linux,x:b2,a:b1,b:b0,y:b3,back:b12,guide:b16,start:b13,dpleft:b10,dpdown:b9,dpright:b11,dpup:b8,leftshoulder:b4,lefttrigger:b6,rightshoulder:b5,righttrigger:b7,leftstick:b14,rightstick:b17,leftx:a0,lefty:a1,rightx:a2,righty:a3,

19000000030000000300000002030000,gameforce_gamepad,platform:Linux,x:b2,a:b1,b:b0,y:b3,back:b8,guide:b16,start:b9,dpleft:b12,dpdown:b11,dpright:b13,dpup:b10,leftshoulder:b4,lefttrigger:b6,rightshoulder:b5,righttrigger:b7,leftstick:b14,rightstick:b15,leftx:a1,lefty:a0,rightx:a3,righty:a2,

05000000c82d00000021000000010000,8BitDo SN30 Pro for Android,platform:Linux,x:b3,a:b0,b:b1,y:b4,back:b10,guide:b12,start:b11,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,dpup:h0.1,leftshoulder:b6,lefttrigger:b8,rightshoulder:b7,righttrigger:b9,leftstick:b13,rightstick:b14,leftx:a0,lefty:a1,rightx:a2,righty:a5,

03000000c82d00000031000011010000,8BitDo 8BitDo Receiver,platform:Linux,x:b3,a:b0,b:b1,y:b4,back:b10,guide:b2,start:b11,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,dpup:h0.1,leftshoulder:b6,lefttrigger:b8,rightshoulder:b7,righttrigger:b9,leftstick:b13,rightstick:b14,leftx:a0,lefty:a1,rightx:a2,righty:a3,

british-choi, 2021-02-13:
@pedropereira1982 
Connect to the ssh terminal and run the following command
# sdljoytest
Then you can get your gamepad information.
make sure the content recorded in gamecontrollerdb.txt matches the gamepad information you found.
I don't have any information about the gamepad you are using

pedropereira1982, 2021-02-13:
> @pedropereira1982
> Connect to the ssh terminal and run the following command
> # sdljoytest
> Then you can get your gamepad information.
> make sure the content recorded in gamecontrollerdb.txt matches the gamepad information you found.
> I don't have any information about the gamepad you are using

Im sorry but i dont know how to connect to ssh terminal

british-choi, 2021-02-13:
@pedropereira1982 
sorry, I cannot help you about ssh terminal.
Search google for ssh connect to EmuELEC

pedropereira1982, 2021-02-13:
> @pedropereira1982
> sorry, I cannot help you about ssh terminal.
> Search google for ssh connect to EmuELEC

Do i need keyboard?
Hope this will be fixed on next update.Theres a lot of people with same problem

shantigilbert, 2021-02-13:
I have not heard anyone else with this issue, so no, there are not a LOT of people with this particular issue (that I know), but either way, I suggest you wait for a proper fix on the image, as you seem to not want/kknow how to follow any suggestions.

pedropereira1982, 2021-02-13:
> I have not heard anyone else with this issue, so no, there are not a LOT of people with this particular issue (that I know), but either way, I suggest you wait for a proper fix on the image, as you seem to not want/kknow how to follow any suggestions.

I apreeciate your patience but i did follow yesterday and today all of the suggestions and none of them work. Except that i dont have keyboard. You can enter facebook on page emuelec and you will see how many posts there are saying the same error. Diferent gamepads, same error.

shantigilbert, 2021-02-13:
I don't use Facebook (nor do I want to). Like I said, if you do not have a keyboard to test the suggestions then just wait for a proper fix on the image, all the suggestions to figure what is going on and do some testing REQUIRE a Keyboard.

british-choi, 2021-02-14:
@pedropereira1982 
You need a keyboard. (USB keyboard or wireless keyboard)
The terminal program can be run from Setup - "13 - Launch Terminal (kb)".

pedropereira1982, 2021-02-14:
> @pedropereira1982
> You need a keyboard. (USB keyboard or wireless keyboard)
> The terminal program can be run from Setup - "13 - Launch Terminal (kb)".

Ok thanks.

shantigilbert, 2021-02-14:
If you only want to install drastic you don't even need to go into the terminal, just use the keyboard to select Yes on the menu

pedropereira1982, 2021-02-14:
> If you only want to install drastic you don't even need to go into the terminal, just use the keyboard to select Yes on the menu

Ok thanks. I will.buy.a keyboard

pedropereira1982, 2021-02-15:
I tried with a keyboard i pressed space and others. It doesnt work

pedropereira1982, 2021-02-15:
I tried another keyboard and works. Just press space. Thank you all. You can close this thread.

closed 2021-02-25 by shantigilbert