EmuELEC/EmuELEC #465: How to add a third party bluetooth controller?

ivan-yu, 2021-02-14:
I have already read the wiki on bluetooth and controllers.
https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth
https://github.com/EmuELEC/EmuELEC/wiki/Recommended-Gamepads

I have a GameSir T4 Pro BT controller.  It supports Nintendo Switch BT mode.  I can confirm it gets paired and trusted with my Odroid N2+ with EmuElec 4.0.  It gets recognized as "Pro Controller".  However, no button has any action at all, so I cannot configure it.

Is there some way to modify or add an entry to joypad configuration so it gets used as Nintendo Pro Controller?  Is it because its GUID is not in the list supported by EmuElec?

Thank you very much for any advice and help!

ivan-yu, 2021-02-15:
Would it be because there is no joypad configuration, like mentioned here?
https://www.retroarch.com/index.php?page=joypad-autoconfig

How can I add my own?

ivan-yu, 2021-02-17:
Does anybody know?  Appreciate any pointer and help, thanks in advance.

shantigilbert, 2021-04-01:
Sorry I do not know, if the device is getting paired but no inputs are being detected you might want to test with `evtest` or `sdljoytest` using SSH and see if its a system wide issue or just on ES.

closed 2021-04-24 by shantigilbert