EmuELEC/EmuELEC #467: EmuELEC 4.0 on OGA: PPSSPPSDL in-game menu hotkey not functioning

soimy, 2021-02-15:
**Describe the bug**
No way to bring out PPSSPPSDL in-game menu with HOTKEY+X(RetroArch way) or Fn5 ( Functioning prior to 3.8).
I know it's better to ask first in discord but all channel is muted.

**To Reproduce**
Steps to reproduce the behavior:
1. launch game using PPSSPPSDL
2. press HOTKEY+X
3. press Fn5

**Expected behavior**
Bring up in-game menu for settings and save states.

**EmuELEC version:**
 - Version 4.0 

**Device 
 - SOC: RK3326
 - Brand: Odroid
 - Model: Odroid Go Advance
 - Version: 1.0

**Additional context**
Confirmed stock `controls.ini` & `ppsspp.ini` in `/storage/.config/ppsspp/PSP/SYSTEM`

shantigilbert, 2021-02-15:
"I know it's better to ask first in discord but all channel is muted."  Read the rules in discord

soimy, 2021-02-15:
[image]
This is all I got. Same messages in history

soimy, 2021-02-15:
Managed to get to the PPSSPP menu using ROM `Jeane d'Arc`'s browse option. Then set `controlmap`.
In case anyone face the same question:
locate `/storage/.config/ppsspp/PSP/SYSTEM/controls.ini`, change the following mapping:
```
Pause = 10-4010
Save State = 10-107
Load State = 10-106
```

in-game menu key: Fn5
Save state: Fn4
Load state: Fn3

closed 2021-02-15 by shantigilbert