EmuELEC/EmuELEC #468: EmuELEC 4.0 on OGA : Wifi lost after power button hibernation

soimy, 2021-02-15:
**Describe the bug**
Running EmuELEC 4.0 on OGA, connecting a working USB WIFI dongle `Brand: Mercury MW150US, chipset: Realtek rtl8188cus`. On power-on everything works flawlessly, but once I push the power button to hibernate the system and turned on again, I lost my wifi connection and no way to restore unless a system reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. Power on OGA device
2. Network settings -> Enable WIFI
3. WIFI connected with proper IP address
4. Press power button to hibernate system
5. Wait a few second and press power button again to light up the system.
6. Goto Network settings -> WIFI not connected
7. re-enable WIFI, search SSID and none is shown

**Expected behavior**
WIFI reconnect after hibernation.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version 4.0

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: RK3326
 - Brand: Odroid
 - Model: Odroid Go Advance
 - Version: 1.0

**Additional context**
USB WIFI dongle info:
`Brand: Mercury MW150US, chipset: Realtek rtl8188cus` https://www.mercusys.com/en/product/details/mw150us
Working properly on first system boot.

shantigilbert, 2021-02-15:
Read the known issues. https://github.com/EmuELEC/EmuELEC/issues/463

Langerz82, 2022-06-07:
https://github.com/EmuELEC/emuelec-emulationstation/pull/59

Andurillek, 2023-03-26:
Same behavior on Odroid Go Super device running EmuELEC 4.6.

shantigilbert, 2025-11-05:
Closing, no reply. If this issue is still present, please reopen