EmuELEC/EmuELEC #470: EmuELEC 4.0 on OGA: Brightness is not saved across reboots

miwasp, 2021-02-15:
**Describe the bug**
No matter what value is set for brightness, it is reset to something different after reboot

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Main menu' (F6)
2. Click on 'System settings'
3. Set brightness to e.g. 81%
4. Reboot
5. Brightness is set back to 31% (in my case)

**Expected behavior**
Brightness should be saved across reboots

**EmuELEC version:**
 - Version 4.0

**Device**
 - Odroid Go Advance Black Edition

shantigilbert, 2021-02-15:
If you set brightness via the hotkey, does it also happen ?

miwasp, 2021-02-15:
In short: Yes

If i change brightness via hotkey, this change can be seen in the system settings. So it makes no difference if you change brightness via hotkey or menu.
The current value is correctly saved in emuelec.conf, but for some reason another value is set after reboot or even after standby.

shantigilbert, 2021-02-15:
Ok thanks, I will look into it.

miwasp, 2021-02-16:
The problem is the different range of "brightness.level" in emuelec.conf (0..100) and "/sys/class/backlight/backlight/brightness" (0..255).

This works so far for me on my OGA, no idea if it works for other devices:

> $ diff /storage/.config/emuelec/scripts/odroidgoa_utils.sh.org /storage/.config/emuelec/scripts/odroidgoa_utils.sh
63,65c63,66
< STEPS="20"
< CURRENTBRIGHT=$(cat /sys/class/backlight/backlight/brightness)
< MAXBRIGHT="255" #$(cat /sys/class/backlight/backlight/max_brightness)
<---
\> STEPS="5"
\> CURBRIGHT=$(cat /sys/class/backlight/backlight/brightness)
\> CURRENTBRIGHT=$(($CURBRIGHT*100/255))
\> MAXBRIGHT="100"
77,78c78,80
< 	echo "${STEPBRIGHT}" > /sys/class/backlight/backlight/brightness
< 	set_ee_setting "brightness.level" $(cat /sys/class/backlight/backlight/brightness)
<---
\> 	NEWVAL=$(($STEPBRIGHT*255/100))
\> 	echo "${NEWVAL}" > /sys/class/backlight/backlight/brightness
\> 	set_ee_setting "brightness.level" $STEPBRIGHT

shantigilbert, 2021-02-16:
Thank you, I will test it out

shantigilbert, 2021-02-17:
I don't understand this part 

```
 	NEWVAL=$(($STEPBRIGHT*255/100))
 	echo "${NEWVAL}" > /sys/class/backlight/backlight/brightness
 	set_ee_setting "brightness.level" $STEPBRIGHT
```

you set one value to the /sys/class, but another to the .conf file? and his works?

shantigilbert, 2021-02-17:
Ok I guess it does, since the value gets converted directly. Thanks for the fix I will push it

miwasp, 2021-02-17:
This works on my OGA.
Today i got my OGS. On OGS i can change the value with the hotkeys and i can see the change in the menu. I can even go to standby and switch it on again and still have the correct value of brightness. But if i restart the OGS, brightness is always 50 and i don‘t know where this is set.

shantigilbert, 2021-02-17:
its set here

```
set_ee_setting "brightness.level" $STEPBRIGHT
```
and its saved on `/emuelec/configs/emuelec.conf`

miwasp, 2021-02-17:
max_brightness on OGA is 255 and i mistakenly assumed it is 255 on all devices.
But max_brightness on OGS is 160!
This works for both OGA and OGS:

>if [ "${1}" == "bright" ]; then
STEPS="5"
CURBRIGHT=$(cat /sys/class/backlight/backlight/brightness)
MAXSYSBRIGHT=$(cat /sys/class/backlight/backlight/max_brightness)
CURRENTBRIGHT=$(($CURBRIGHT*100/$MAXSYSBRIGHT))
MAXBRIGHT="100"
MINBRIGHT="2"
        if [ "${2}" == "+" ]; then
                STEPBRIGHT=$(($CURRENTBRIGHT+$STEPS))
        elif [ "${2}" == "-" ]; then
                STEPBRIGHT=$(($CURRENTBRIGHT-$STEPS))
        else
                STEPBRIGHT=${2}
        fi
        [ "$STEPBRIGHT" -ge "$MAXBRIGHT" ] && STEPBRIGHT="$MAXBRIGHT"
        [ "$STEPBRIGHT" -le "$MINBRIGHT" ] && STEPBRIGHT="$MINBRIGHT"
        #echo "Setting bright to $STEPBRIGHT"
        NEWVAL=$(($STEPBRIGHT*$MAXSYSBRIGHT/100))
        echo "${NEWVAL}" > /sys/class/backlight/backlight/brightness
        set_ee_setting "brightness.level" $STEPBRIGHT
fi

shantigilbert, 2021-02-17:
If you wish you can open a PR, or I can push the fix, let me know.

miwasp, 2021-02-17:
Please push the fix...

commit 288ba9f3 (2021-02-17): Fix brightness not restored after reboot on OGA/S

Thanks to @miwasp, fixes #470
closed 2021-04-24 by shantigilbert