EmuELEC/EmuELEC #473: EMUELEC 4.0 on ODROID N2+

knutzone, 2021-02-16:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

knutzone, 2021-02-16:
Hi there,

I installed the new EmuElec 4.0 on my Odroid N2+ and now i have some questions about it:

1.) It should be possible to have dolphin on the N2 (S922x device) , but I can’t see it anywhere! I tried to add the “dolphin-data.zip” in Retroarch (Main Menu/Online Updater/Content Downloader/Nintendo-GameCube-WII), but nothing happens after that. I’m not able to find the Dolphin Core or something like e.g. “GameCube”. In Emulationstation the same, nothing appears there.

2.) And, if it is present, where can I place my GameCube-roms? I even can’t find a folder for it! Do I have to create it manually? And how is it’s name? 

3.) In previous versions of EmuElec I had force feedback (vibration) on my Logitech Cordless Rumblepad 2.0 in some games (e.g. Rush 2049), but now it’s totally gone! I also could set the feedback on Retroarch (Settings/Input/Haptic Feedback/Vibration), but now it only shows “No Settings Found”.

4.) In Emulationstation the configuring of my controller Logitech Cordless Rumblepad 2.0 is not working in the analog section! Both analog sticks are not recognized. For a working setup I have to do this in Retroarch, then it works even under Emulationstation. This behaviour appeared some versions earlier, I think since EmuElec 3.6 or so. Before it worked without problems.

Ok, that’s it for the moment. I hope that there are some solutions for my requests...
And please don’t think that I only want to criticize the project – no, of course not, You are doing a very great job and many thanks for all of your phantastic work!

AlienatedDiego, 2021-02-17:
1. Dolphin is the standalone version, not a libretro_core, that's why it's not there.
2. "gamecube" / "wii" folder inside roms directory (roms / gamecube)
3 & 4. I don't know the reason, it must be checked. Thank you for reporting.

knutzone, 2021-02-17:
Thanks for your reply! 
Ok., I can't find ay folder named "gamecube" in my roms folder. Do I have to self-create it?
If I open a gamecube content inside Retroarch there's no possibility to open it with dolphin core.
Thank you for caring!

AlienatedDiego, 2021-02-17:
Yes, you must create the gamecube folder in the roms directory and put some iso so that the system is visible in E.S
Auto controller mapping is not possible, this will be included in next release 4.1
You can try to map manually by editing the GCPadNew.ini file (storage / .confif / emuelec / configs / dolphin-emu)

knutzone, 2021-02-17:
I tried to create the folder, but it replies "Failure" with error-code 4. What does this mean?

AlienatedDiego, 2021-02-17:
Idk, not sure, I should be able to create roms / gamecube folder and enter isos without problem. Note that I don't use image for N2, but generic amlogic ng (s905x3). I don't know if it should make a difference. how are you accessing roms directory ?. Try by network, shared folder "emuelec"

AlienatedDiego, 2021-02-17:
If the system is installed in SD you can also insert it into windows pc to access the EEROMS partition

shantigilbert, 2021-02-17:
@knutzone Please do not mix issues, open a report for each one.

And there has been no changes to how controllers work AFAIK, so who knows why its happening, I do not own the controller and unless someone goes in and debugs it I have no real idea.

as far as Gamecube/wii goes, there is no Retroarch core and you need to do a lot of manual work to get it to work, so I think you should wait until a release has it fully implemented, if ever (Dolphin is not that great on EmuELEC, even on the N2+)

shantigilbert, 2021-02-25:
No reply for 8 days, closing

closed 2021-02-25 by shantigilbert