EmuELEC/EmuELEC #475: EmuELEC 4.0 on Odroid Super : Issue with RetroAchevements password in RetroArch when using ee_defaults.txt

Desani, 2021-02-17:
I am using the ee_defaults.txt file to store login information for RetroAchievements 

In Emuelec menu's, RetroAchievements work fine. I click on RetroAchievements on the Emuelec screen, it logs in without issue and I can see my achievements

In each game I launch through retroarch gives me the following error:

RetroAchievements: Error with login. Please try again.

I think I found the culprit for the Retroarch RA error. When looking into the retroarch.cfg, at the cheevos_password = field, it looks like it inserts a new line character at the end of the password. So it is attempting to authenticate with a password with a new line character inserted.

shantigilbert, 2021-02-17:
Yes this has been fixed in dev. meanwhile use a text editor that uses Linux end lines and make sure you have no spaces at the end of the passwords.

Desani, 2021-02-17:
Okay sounds good. I was just reviewing the known bugs for 4.0 release and I did not see it mentioned, so I thought I would open a bug report.

Thanks for the fix.

closed 2021-04-24 by shantigilbert