EmuELEC/EmuELEC #477: Micro SD card problem after writing image with ETCHER

drewsie001, 2021-02-18:
Hi there. Really sorry to be posting this as it's my first ever post on here.

I've looked on the wiki page and exhausted every Google search and I'm still struggling to find a solution to my problem.

With Etcher, I'm writing the EmuELEC image to a 64GB SanDisk sd card with adapter. I've also used a USB 64GB SanDisk card but exactly the same problem.

Everything seems fine and I can replace the dtb file, no issues. The problem I have is the Memory card goes from 64gb down to 512mb card?????? 
I then have to format the card/s in my laptop but when the formatting is done and I'm still left with only 512mb. I then have to plug the micro sd card into my tablet and format it so that it's back to its original state.

I've followed YouTube videos to the letter and everything I'm doing is ok. I've tried the process on 2 different laptops and on a mac book pro but everytime I get the same results. Its absolutely baffling. 

My set up is 

T95 Max Android TV box 4GB Ram 32GB Rom
SanDisk Extreme 64gb + SanDisk Adapter
SanDisk 64gb USB (think it's called a Cruzer)

Really sorry for a lengthy first post but I'm literally pulling my hair out).

Thanks in advance. Any help/tips will be greatly appreciated

shantigilbert, 2021-02-18:
Did you by any chance missed the part where you need to use the card on the device right after flashing it to get the partitions resized?

drewsie001, 2021-02-18:
mmnmmmm.....the videos I've watched shows after flashing the card, you remove it then plug it straight back into the PC, then replace the dtb file. It's at this point that the card has shrunk down to 512mb.

Are you saying that I have to flash the card, then pop it into the Android box before replacing the dtb file?

dm2912, 2021-02-18:
no, once you have added the dtb then you need to boot it up on your android device. then the partitions will be created.

How do you expect etcher to magically know what partitions need creating, how big they should be? The only way to do that would emuelec being shipped in 32gb, 64gb,m 128gb etc images..

when you first boot to emuelec, the installer will create the partitions

drewsie001, 2021-02-18:
Yes I've done that, that's how I first knew I had an issue. After the first boot is complete i then start adjusting the settings, then I start adding the roms. It's at this point I first encountered the problem because the space on the card just wasn't there.

I took the card out of the box and popped it back into my laptop and it was telling me my sd card was  now only 512mb

This is the process that I follow....

Put sd card in laptop
Start up Etcher
Select the file
Choose the correct drive to flash to
And then flash the image

I then take out the card, pop it back in (because its not showing up as a recognised drive) and replace the dtb file. 

I've since noticed, before I even eject the card from the laptop that the sd card has already been altered to 512mb.

If I'm doing something glaringly obvious please forgive me but as I say I've followed numerous YouTube videos to the letter.

I truly thank you for your patience

drewsie001, 2021-02-18:
After rereading my original post it does look as though I've had misled you by not confirming that after replacing the dtb file I haven't put the card in my  device. Just to be clear, I have. That's how I first learnt of the issue. The space on the card so ran out after I started adding roms to the card.

dm2912, 2021-02-18:
When you first burn the image, yes, there will only be two partitions, and one will be very small with a placeholder "resize_me" file.. the proper partitions are not created until the system installs and boots for the first time

There should be 3 partitions

The main emuelec partition which is 2gb
The storage partition which is 2gb
And then the remaining space will become the EE_Roms partition which is fat 32 and is where your roms go

how many partitions do you have? Which one are you saying is only 512mb? The EE_Roms partition?

If the fat32 partition is only 512mb then something has gone wrng with the install and you likely need to reflash and go again.

I should probably ask at this point, is your SD Card genuine????

You dont need youtube videos, they arent always right, and unless they are for 4.0 they don't apply anyway because the partition layout changed for 4.0

Flash to the card
Add the dtb (If you do remove the card make sure you use the safely remove controls to ensure its unmounted before removing)
Put it in the box
allow the install process to run
boot into emuelec
at this point all three partitions should have been created.

drewsie001, 2021-02-18:
I really appreciate the responses and I also appreciate the patience that is being shown. Yes dm you're probably right, I am getting hung up on the fact that I'm seeing 512mb left on a 64gb  card. I've been at it for the last 2 days and I've been ripping my hair out.

I'm going to have a another go in a short while and I'll report back.

Cheers for now

shantigilbert, 2021-02-18:
512mb on the partition means this was prior to v4.0, which means the second partition - unless you use Linux - will not be mounted. 

As Dim mentions, v4.0 changed partition layout, so 

Partiotion 1 and 2 should be 2gb
Partition 3 will get resized to whatever is left on the card

But only Partition 1 and 3 are visible in Windows 

Partitioning happens after first boot, so after flashing and dtb renaming.

Destes099, 2021-02-19:
I had the same issue on my laptop. Are you getting a error msg after you flash the image? I had to use rufus for it to work on my laptop.

drewsie001, 2021-02-19:
Thanks guys everything is all set up now and ready to go.

I was totally unaware that when you put the SD card into the device the initial boot does creates all the partitions etc. 

Really appreciated all the help and advise guys. Thank you so much.

closed 2021-02-21 by drewsie001

yunnysunny, 2021-12-12:
It seemed emuelec remove the `sd_mod` driver. When I insert my usb disk, it can not been recognised.

When I input the command `modprobe sd_mod`, it will show error as some so file not found.

```
modprobe: FATAL: Module sd_mode not found in directory /lib/modules/3.14.29
```