EmuELEC/EmuELEC #479: Emuelec 4.0 s905x3 4gb ram scraper not working!!

Destes099, 2021-02-19:
I downloaded the new 4.0 software. I reinstalled all of my ROMs. Everything was going smoothly and good still trying to add additional ROMs to complete my collection. When I go install a new rom I then click the select button go down to game options then I go down to scrape it pulls up two different box arts. I then go to select the one that I want. Then it takes me back out to the screen where you select the games. But there is no box art. If you go back in and look at the game's metadata it shows all the info in there but no box art. Please fix!!!!

dm2912, 2021-02-19:
reload gamelists after scraping

main menu > game settings > update game lists

Destes099, 2021-02-19:
I tried that. Doesn't do anything.

Destes099, 2021-02-19:
I even tried restarting the whole system and box. But it still does nothing. On version 3.9 all I had to do was scrape it would pull up the box art then I would push save at the bottom and voila it was on there.

Destes099, 2021-02-19:
[image]

Destes099, 2021-02-19:
[image]

AFTER PUSHING ACCEPT RESULTS

Destes099, 2021-02-19:
[image]

And once again after updating games list.

Destes099, 2021-02-19:
[image]
All the information is filled in after scraping.

Destes099, 2021-02-19:
But no box art!!

Destes099, 2021-02-19:
Also whenever I have the previous game aren't set to show a little preview of the game. None of them are showing so when I go into the metadata and select the box art for the preview it will show. Then when when I update games list it then takes the previews away.

dm2912, 2021-02-19:
it literally says on your screenshot that it does not have a logo/box/titleshot so there is nothing to show. If the box doesnt exist it cannae show it

what do you have for scraper settings? Do you have all those sources enabled?

Are you logged in to a screenscraper account?

Also, screenscraper put out an announcement last night that they were having issues

[image]

Destes099, 2021-02-19:
All the other games that I have that have box arts on them and they dont have anything in the logo/box/titleshot. Also I do not have a login to screen scraper as I did not on the last version and it worked just fine. Also they added two new box art collectors on this version. But they are not working also.

Destes099, 2021-02-19:
Also I was able to add box art manually before as well and now this is what it says when I try to do that

Destes099, 2021-02-19:
[image]

Destes099, 2021-02-19:
[image]

Destes099, 2021-02-19:
It says the disk write is protected.

dm2912, 2021-02-19:
> All the other games that I have that have box arts on them and they dont have anything in the logo/box/titleshot. Also I do not have a login to screen scraper as I did not on the last version and it [...]

its not version related, screenscraper closed their API in late december to members only, they only reopened it today or so. 

> It says the disk write is protected.

that generally happens when you have access the storage and not done a safe eject - so either you took it out of your computer without ejecting it (linux) or clicking safely remove (windows) - or you shut down your box by hard reset, ie turning the power off. That causes partitions to lock themselves as read only.

Put the storage back in your device, reboot properly a few times via the menu and eventually the read only should lift.

dm2912, 2021-02-19:
> All the other games that I have that have box arts on them and they dont have anything in the logo/box/titleshot. Also I do not have a login to screen scraper as I did not on the last version and it [...]

if they dont have anything for boxart in the metadata they 100% cannot display a boxart. thats simply impossible

what is likely is that you downloaded box art to "image" rather than boxart.. and changed the scraper settings.

I did ask what your scraper settings were

dm2912, 2021-02-19:
> [image]
> All the information is filled in after scraping.

no it isnt. logo, box and title shot are not filled in.

Destes099, 2021-02-20:
The desk lock went away and for whatever reason the screen scraper just started working. Screenshot I sent to you was accessing ROMs over IP address.

dm2912, 2021-02-20:
> The desk lock went away and for whatever reason the screen scraper just started working. Screenshot I sent to you was accessing ROMs over IP address.

if the drive was locked that is why the scraper could not download and add the artwork.

if a drive becomes locked its 99/100 because

a) the device was hard reset or shut down direct at the power point.; linux especially likes to mount and unmount drives properly, if you shut down without unmounting for whatever reason the drive will lock as read only because it things its mounted by another system

b) the usb or card being removed from a windows (or linux) machine without doing the safely eject removable media routine. again, it leads the drive to believe its still mounted so prevents further access

a few proper reboots usually sorts it out. it went away for whatever reason presumably because you either rebooted or did a safe unmount.

closed 2021-02-22 by shantigilbert