EmuELEC/EmuELEC #482: EmuELEC 4.0 Test on Odroid Super : Issue scanning for new ScummVM Games

Desani, 2021-02-19:
**Describe the bug**
Launching 16 - Scan ScummVM Games in the setup menu just exits to the menu.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Setup Menu
2. Click on Scan ScummVM Game
3. There is no script output, it just reloads the menu

**Expected behavior**
Script is supposed to launch /usr/bin/scummvm.start but it is not marked as executable and so it fails to run. 

**EmuELEC version:**
 - Version 4.0 Test that was just released. (Same issue on 4.0 release)

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - Brand: Odroid
 - Model Super

**Additional context**
I can't mark the script as executable as it is mounted as read only to test if that fixes the issue.

shantigilbert, 2021-02-19:
https://github.com/EmuELEC/EmuELEC/issues/463

Desani, 2021-02-19:
You mention that it is a known issue, and I read before and did not open a bug report for it but you released a test version today and announced it on discord with the following fixes:

"I just uploaded a test release, hopefully the auto update works, but you need to set it to "beta", if it doesn't please let me know! If it does not detect it you can also use the "Force update" on the danger zone, but make sure you set it to "beta" before
* Fix Scummvm game scan"

I updated to that test release this morning and still had this issue, which is why I opened a bugfix because it is marked as fixed but I still had an issue.

shantigilbert, 2021-02-19:
did you update correctly ? 

https://github.com/EmuELEC/EmuELEC/wiki/How-to-update

specifically this: 
`Navigate to "EmuELEC settings" -> "Danger Zone" -> "RESET EMUELEC SCRIPTS AND BINARIES TO DEFAULT"`

Desani, 2021-02-19:
I did not update correctly. I will attempt to do that now.

Desani, 2021-02-19:
Updated using the correct method and I was able to scan and add scummvm games correctly. Sorry for the inconvenience

shantigilbert, 2021-02-19:
Can you close this issue please

closed 2021-02-19 by shantigilbert