EmuELEC/EmuELEC #484: Build 4.0 fail - Folder /tmp/tmp.T4gRvNNFPz might be out of free space

keeno1982, 2021-02-20:
Running /home/dietpi/emu/projects/Amlogic/bootloader/release

image: creating file EmuELEC-Amlogic.aarch64-4.0-LePotato.img...
image: creating msdos partition table...
image: creating part1...
image: creating part2...
image: creating filesystem on part1...
image: creating bootloader configuration...
Writing u-boot to EmuELEC-Amlogic.aarch64-4.0-LePotato.img
image: copying files to part1...
image: extracting part2 from image...
image: creating filesystem on part2...
image: populating filesystem on part2...
image: merging part2 back to image...
image: extracting part1 from image...
image: An error has occurred...

Folder /tmp/tmp.T4gRvNNFPz might be out of free space...

image: cleanup...

make: *** [Makefile:12: image] Error 1

shantigilbert, 2021-02-22:
"might be out of free space"

closed 2021-02-22 by shantigilbert

keeno1982, 2021-02-23:
@shantigilbert  I have 285G left, so I don't think it really "out of free space".

shantigilbert, 2021-02-23:
No idea then, but I do know this on your side, as I just finished a clean build and had no issues. Sorry

keeno1982, 2021-06-03:
@shantigilbert What's wrong? 
Filesystem               Size  Used Avail Use% Mounted on
udev                     2.0G     0  2.0G   0% /dev
tmpfs                    395M   13M  382M   4% /run
/dev/sda1                473G  105G  349G  24% /
tmpfs                    2.0G     0  2.0G   0% /dev/shm
tmpfs                    5.0M     0  5.0M   0% /run/lock
tmpfs                    2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs                    2.0G     0  2.0G   0% /tmp
tmpfs                     50M   20K   50M   1% /var/log

keeno1982, 2021-06-03:
Creating 4.0 filesystem on /home/dietpi/emu/target/EmuELEC-Amlogic.aarch64-4.1.system, block size 524288.
[=========================================================================================================================================================================================-] 14787/14787 100%

Exportable Squashfs 4.0 filesystem, lzo compressed, data block size 524288
        compressed data, compressed metadata, compressed fragments, no xattrs
        duplicates are removed
Filesystem size 671744.23 Kbytes (656.00 Mbytes)
        49.68% of uncompressed filesystem size (1352170.42 Kbytes)
Inode table size 185373 bytes (181.03 Kbytes)
        37.10% of uncompressed inode table size (499679 bytes)
Directory table size 169441 bytes (165.47 Kbytes)
        50.64% of uncompressed directory table size (334603 bytes)
Number of duplicate files found 815
Number of inodes 15063
Number of files 13128
Number of fragments 926
Number of symbolic links  1029
Number of device nodes 0
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 906
Number of ids (unique uids + gids) 3
Number of uids 2
        root (0)
        dietpi (1000)
Number of gids 3
        root (0)
        unknown (81)
        dietpi (1000)
Running /home/dietpi/emu/projects/Amlogic/bootloader/release

image: creating file EmuELEC-Amlogic.aarch64-4.1-LePotato.img...
image: creating msdos partition table...
image: creating part1...
image: creating part2...
image: creating filesystem on part1...
image: creating bootloader configuration...
Writing u-boot to EmuELEC-Amlogic.aarch64-4.1-LePotato.img
image: copying files to part1...
image: extracting part2 from image...
image: creating filesystem on part2...
image: populating filesystem on part2...
image: merging part2 back to image...
image: extracting part1 from image...
image: An error has occurred...

Folder /tmp/tmp.FPwEBCq7jg might be out of free space...

image: cleanup...

make: *** [Makefile:12: image] Error 1

shantigilbert, 2021-06-03:
It still says you are out of free space 2gb is certainly NOT 285gb of free space