EmuELEC/EmuELEC #485: Bluetooth is not working

hobapli, 2021-02-20:
My device is 96mini 2g/16g (gxl_p281_2g.dtb)

I have several problems especially Bluetooth. It's working if boot from nand but cant connect my ds4 on ES. 
Another problem is can't list themes when i start themes menu(waiting and nothing happens). Also file browset is crashing when its starts...

SamuEDL, 2021-02-20:
Good Morning.
What bluetooth chip does this tvbox have? Without that information, we cannot help you.

hobapli, 2021-02-20:
> Good Morning.
> What bluetooth chip does this tvbox have? Without that information, we cannot help you.

thank you for quick reply, how may i know that? Aida64 doesn't show me the chip model...

hobapli, 2021-02-20:
damn please forgive me. Confused with my other mi box device, so this device has no bluetooth lol

SamuEDL, 2021-02-20:
> damn please forgive me. Confused with my other mi box device, so this device has no bluetooth lol

So if the device doesn't have bluetooth. I proceed to close the problem.

closed 2021-02-20 by SamuEDL