EmuELEC/EmuELEC #487: EmuELEC 4.0 on Odroid Advance 1.0：can not CPU Overclock

lancherliu, 2021-02-21:
Describe the bug
can not CPU Overclock

To Reproduce
Steps to reproduce the behavior:
1. Go to 'EMUELEC SETTINGS'-'DANGER ZONE'-
2. Click on 'OVERCLOCK'
3. Set  'OVERCLOCK' to 1.5GHZ
4. In SSH the CPU stats CPU Freq  is also 1296000

Screenshots
[image]
[image]

**EmuELEC version:**
 - Version 4.0

Device
Odroid Go Advance 1.0

shantigilbert, 2021-02-23:
This has been fixed on v4.1 test/beta

* Fixed issue with OGA/S OC not beeing applied correctly (this does not solve the random lockups on some devices)

closed 2021-04-24 by shantigilbert

Lyteral, 2021-08-31:
From what I can tell, the Overclock on Odroid Go Advance 1.0 is still not applied on most current version 4.3-TEST

[image]
 
Was set up via UI to 1.4 GHz