EmuELEC/EmuELEC #489: Can't build Emuelec 4

dovart, 2021-02-22:
**Describe the bug**
There is no way to build Emuelec 4 from sources.

SDL not found at mirror:
https://sources.libreelec.tv/mirror/

**https://sources.libreelec.tv/mirror/SDL/SDL-74055dd839e4.tar.gz**
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
**HTTP request sent, awaiting response... 404 Not Found**
**Expected behavior**

--2021-02-22 00:41:38--  http://sources.libreelec.tv/mirror/SDL/SDL-74055dd839e4.tar.gz
Resolving sources.libreelec.tv (sources.libreelec.tv)... 138.68.75.163
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://sources.libreelec.tv/mirror/SDL/SDL-74055dd839e4.tar.gz [following]
--2021-02-22 00:41:39--  https://sources.libreelec.tv/mirror/SDL/SDL-74055dd839e4.tar.gz
Connecting to sources.libreelec.tv (sources.libreelec.tv)|138.68.75.163|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-02-22 00:41:40 ERROR 404: Not Found.

keeno1982, 2021-02-22:
modify ./packages/sx05re/tools/SDL/package.mk and remove PKG_SHA256 value.

shantigilbert, 2021-02-22:
https://github.com/EmuELEC/EmuELEC/commit/0ec9e9acd2e672e7bf63025beec1906867247239

closed 2021-02-22 by shantigilbert