EmuELEC/EmuELEC #490: EmuELEC 4.0 - DraStic settings

Owtinho, 2021-02-22:
Hello, 
I installed the new version from EmuELEC and everything work fine. But I don't know how I can change the settings from DraStic for the key assignment or that I see only one screen and not both at the same time.

Thank you for an answer.

AlienatedDiego, 2021-02-22:
Please follow the template or provide the minimum necessary information of the device and version of EmuELEC

AlienatedDiego, 2021-02-22:
Drastic by default is configured to display both screens as in the original system. Where did you get that Emuelec 4.0?

Owtinho, 2021-02-22:
I installed it on my rk2020. I tried all the buttons to get into the menu without success. I found out that I can close the game with start and select.

SamuEDL, 2021-02-22:
We do not support rk2020.

closed 2021-02-22 by SamuEDL

Owtinho, 2021-02-22:
That's sad. Maybe it is possible to change the button layout in a file?