EmuELEC/EmuELEC #493: Game shortcuts

durstboiler, 2021-02-23:
Sorry to post here, but I was unable to write a message on Discord.

**Is your feature request related to a problem? Please describe.**
I cannot create several configuration (widescreen, test, HD ect) for the same N64 games, without duplicating the rom.

**Describe the solution you'd like**
I would like to be able to create game shortcuts to have multiple configurations for the same game. It works with m3u on PSX and DC, but not for N64. 

**Describe alternatives you've considered**
...

**Additional context**
...

dm2912, 2021-02-23:
i think i know what you mean, i guess you could create simlinks

so if you have mario_kart.n64

you could create mario_kart2.n64 as a simlink to the first, but im not sure how drive space would be reported.. i think it would still treat the linked file as the same size as the original

shantigilbert, 2021-02-23:
As is I think this is currently not possible, not without copying the ROMS

closed 2021-02-24 by shantigilbert